linux4wilc/driver #71: Concurrency STA and AP

bht-patrick-schneider, 2019-10-01:
Hi,
I have a question about concurrency in the ATWILC3000 module.
This document states http://ww1.microchip.com/downloads/en/DeviceDoc/Wi-Fi%20Link%20Controller%20Linux%20Driver%20Release%20Notes%20v15.2.pdf
that concurrency modes in multichannel are not supported.
Our use case would be to open an Access Point with hostapd on a fixed channel and connect to user wifi networks with wpa_supplicant if one becomes available (channel not known).
Is it somehow possible to surpass the limitians for our use case scenario?

Thanks for your help!
Best regards, 
Patrick

ShonkyCH, 2019-10-01:
If you could work around it, then it wouldn't really be a limitation.

https://github.com/linux4wilc/driver/issues/43

bht-patrick-schneider, 2019-10-01:
Thank you for pointing me to that issue for clarification.
So our use case can not be solved that easy.
Is it a module limitation by hardware or software (and therefore possibly lifted in future releases?).

closed 2019-10-29 by AdhamAbozaeid