linux4wilc/driver #75: Wifi AP won't connect dev branch

jackmitch, 2019-10-21:
Hi,

On the latest dev branch with matching firmware + kernel v5.3 my wifi AP won't connect. This works OK on the master branch. It seems to hang for ~10 second at sending packet.

[bad_log.txt](https://github.com/linux4wilc/driver/files/3750267/bad_log.txt)

jackmitch, 2019-10-21:
Sorry, this this was a configuration issue on my end.

closed 2019-10-21 by jackmitch

AdhamAbozaeid, 2019-10-21:
Thanks @jackmitch for the update

jackmitch, 2019-10-21:
I did see some more issues with it after I closed this, I think it's something around the authentication side, my Android 10 phone seems to have issues with it, but my Linux desktop is OK. It's going to go through some more testing in the next couple of days and I'll update.