linux4wilc/driver #76: kernel NULL pointer at  wilc_wlan_txq_remove_from_head+0x58/0xd4

mgkiller7, 2019-10-29:
kernel NULL pointer at  wilc_wlan_txq_remove_from_head+0x58/0xd4， which cause hang while doing  iper3.

I use WILC1000-SD wifi module insert to NK-NUC980 EVK Board SD1 slot(新唐) uesing this github driver. 

Start WILC1000-SD wifi（STATION mode） log：
~ # sh /nvm/wifi.sh 
Successfully initialized wpa_supplicant
[   78.980599] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   78.992274] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   79.003549] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   79.013166] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   79.024233] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   79.033816] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   79.044883] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
rfkill: Cannot o[   79.056658] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[c399a000] wlan0
pen RFKILL control device
[   79.066908] WILC POWER UP
[   79.070474] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_init_host_int]Host[c399a000][c399b428]
[   79.078724] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[   79.087874] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[   79.095749] wilc_sdio mmc0:0001:1: SDIO speed: 20000000
[   79.103074] wilc_sdio mmc0:0001:1: chipid 001003a0
[   79.109108] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
[   79.119141] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
[   79.127974] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
[   79.139491] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
[   79.149049] wilc_sdio mmc0:0001:1: failed to get IRQ GPIO, load default
[   79.157458] wilc_sdio mmc0:0001:1 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 347
[   79.166266] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC1000
[   79.176366] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000000
[   79.182449] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc1000_wifi_firmware.bin
[   79.194758] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc1000_wifi_firmware.bin
[   79.207416] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
[   79.217708] wilc_wlan_firmware_downloadDownloading firmware size = 135060
[   79.246708] wilc_wlan_firmware_download Offset = 119744
[   79.254924] wilc_wlan_firmware_download Offset = 135060
[   79.261724] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_firmware_download]Download Succeeded
[   79.270049] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_start_firmware]Starting Firmware ...
[   79.278949] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_start_firmware]Waiting for FW to get ready ...
[   79.363808] cfg_indicate_rx: Info message received
[   79.368799] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_start_firmware]Firmware successfully started
[   79.387866] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Firmware Ver = WILC_WIFI_FW_REL_15_2_1 Build: 11257
[   79.400216] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_init_fw_config]Start configuring Firmware
[   79.503424] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]Mac address: fa:f0:05:9a:88:b6
[   79.524933] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 0.
[   79.534616] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_set_multicast_list]Enable mcast filter retrive directed pkts only
[   79.548408] wilc_sdio mmc0:0001:1 wlan0: INFO [set_power_mgmt]dev [wlan0]
[   79.555333] wilc_sdio mmc0:0001:1 wlan0: INFO [set_power_mgmt] Power save Enabled= 0 , TimeOut = -1
[   79.580524] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 1.
[   79.590208] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 01:00:5e:00:00:01
[   79.641583] wilc_sdio mmc0:0001:1 wlan0: INFO [flush_pmksa]Flushing  PMKID key values
[   79.656558] random: wpa_supplicant: uninitialized urandom read (32 bytes read, 125 bits of entropy available)
Selected interface 'wlan0'
[   81.994874] wilc_sdio mmc0:0001:1 wlan0: INFO [scan]Requested num of channel 14
OK
[   82.010749] wilc_sdio mmc0:0001:1 wlan0: INFO [scan]Scan Request IE len =  0
[   82.017783] wilc_sdio mmc0:0001:1 wlan0: INFO [scan]Number of SSIDs 1
[   82.039499] wilc_sdio mmc0:0001:1 wlan0: INFO [scan]Trigger Scan Request
Selected interface 'wlan0'
bssid / frequency / signal level / flags / ssid
Selected interface 'wlan0'
OK
Selected interface 'wlan0'
0
Selected interface 'wlan0'
OK
Selected interface 'wlan0'
OK
[   82.251866] random: nonblocking pool is initialized
Selected interface 'wlan0'
[... 119 lines cut ...]
[  995.136816] Code: e1530002 0a00001e e3a04c02 e3a05c01 (e5932004) 
[  995.142891] ---[ end trace 009a64cf14a0d9c1 ]---
[  995.147466] note: K_TXQ_TASK[2797] exited with preempt_count 1
[  995.155749] cfg_indicate_rx: Receive unknown message 255-52-250-64-12-83-72-52
[  995.163099] cfg_indicate_rx: Receive unknown message 255-255-255-52-250-64-12-83
[  995.170549] wilc_wlan_handle_rx_buff: Data corrupted 0, 0
[  995.176258] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x0000003f
[  4]   4.00-5.11   sec   102 KBytes   752 Kbits/sec    5   1.41 KBytes       
[  4]   5.11-6.00   sec  0.00 Bytes  0.00 bits/sec    1   1.41 KBytes       
[  4]   6.00-7.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]   7.00-8.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes       
[  4]   8.00-9.00   sec  0.00 Bytes  0.00 bits/sec    0   1.41 KBytes   
-----to here the iperf3 is hang，except "Ctrl+C"---------------

After  "Ctrl+C" log out:
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bandwidth       Retr
[  4]   0.00-107.39 sec   240 KBytes  18.3 Kbits/sec    6             sender
[  4]   0.00-107.39 sec  0.00 Bytes  0.00 bits/sec                  receiver
iperf3: interrupt - the client has terminated

My Quentions are:
1、How to deal with  kernel NULL pointer？
2、Why the Bandwidth of iperf3 test is so small？
     （I have limited the SD_clk to 20MHz, becaue 25MHz ~50MHz is not work well）

AdhamAbozaeid, 2019-10-29:
Hi @mgkiller7 
the UNKNOWN_INTERRUPT logs you are getting are very suspicious. I'm getting much better throughput with iperf 2, but I don't suspect that it will differ much if I used iperf 3.

Please report this issue on [Microchip's SalesForce](http://microchipsupport.force.com/)

mgkiller7, 2019-10-30:
Hi, AdhamAbozaeid
can you get me a exact link to post this issue.   The link you post above make me confuse.

ShonkyCH, 2019-10-30:
> Hi, AdhamAbozaeid
> can you get me a exact link to post this issue. The link you post above make me confuse.

https://microchipsupport.force.com/s/article/How-to-submit-a-case

ruschigo, 2020-04-01:
Hi, @mgkiller7!
do you find any solution? i have similar issue...

Thanks,

Best regards,

Igor Ruschi

mgkiller7, 2020-04-02:
Hi, Igor Ruschi
      Sync your wilc driver&firmware to the latest version from github and have a try again. 
      Good luck to you.
Best regards,
mgkiller7