linux4wilc/driver #77: ATWILC3000 shield with CM3+ and IO board

dicrescenzo3, 2019-10-31:
Hi all,

I'm trying to make the ATWILC3000 shield board working with a CM3+ with IO board.

I'm using the drivers from the Microchip repo:
https://github.com/linux4wilc/driver/

I made a custom overlay as shown below:
`// Overlay for the Microchip ATWILC3000 WiFi Shield
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835";

	fragment@0 {
		target = <&spi0>;
		__overlay__ {
			/* needed to avoid dtc warning */
			#address-cells = <1>;
			#size-cells = <0>;

			status = "okay";

			atwilc3000: atwilc3000@0{
				compatible = "microchip,wilc1000", "microchip,wilc3000";
				reg = <0>; /* CE0 */
				#spi-max-frequency = <12000000>; //LDCT: original
				spi-max-frequency = <1000000>; //LDCT: modified
				reset-gpios = <&gpio 27 0>;
				chip_en-gpios = <&gpio 22 0>;
				irq-gpios = <&gpio 17 0>;
				status = "okay";
			};
		};
	};

	fragment@1 {
		target = <&spidev0>;
		__overlay__ {
			status = "disabled";
		};
	};
};`

I downloaded the linux kernel from the RPi repo, added source codes in the appropriate directory, configured the kernel with menuconfig. Everything works almost fine. But I have two problems:

1) Upon a system reboot I need to rmmod + insmod the wilc_spi.ko module to make the device connecting properly to an AP. If I don't do this the WiFi icon keeps blinking but never reaches a stable point. With rmmod + insmod instead I can connect, disconnect and re-connect correctly. What could be the issue here?

2) The driver needs 3 pins to properly control the module: reset-gpios, chip_en-gpios and irq-gpios. If I omit them from the overlay then the driver falls back into hardcoded values (that I modified) and from the scope I see "good" activity across all the lines (reset for 5ms, clock miso and mosi toggling). Instead, if I pass the gpios from the overlay, as shown above, the driver recognizes the pins but then fails to initialize the module. The scope shows good activity on SPI lines (except MISO) but no activity on reset-gpios, chip_en-gpios and irq-gpios lines. The driver seems to configure gpios direction correctly (return values of gpiod_direction_input/outout are ok) but actually it cannot work with them all. Also, if I check pin configuration with gpio readall it clearly shows that reset-gpios, chip_en-gpios and irq-gpios still have default values. Is the pins declaration in the overlay correct?

Thank you!
LDCT

mmader87, 2019-11-11:
I had a similar problem CM3+ IO board with Yocto Linux Kernel v5.3.  In the driver for the WILC, file wilc_netdev.c and function wilc_wlan_power -> there is a call to gpiod_put(gpio_chip) and gpiod_put(gpio_reset).  This will release the GPIO back and essentially resets those pins to the default value.  This causes the output of those signals to do weird things.  I ended up adding the variables gpio_reset and gpio_chip_en to the wilc struct and removing the respective gpiod_put function calls.

Mateusz-Gwara, 2019-11-14:
Thanks for the inspiration @dicrescenzo3 !
Your first point is comparable to my experience on a AM3352 CPU.
Basically after booting and starting an AP on the WILC3000, any client connecting to it would encounter connectivity problems until I restart the access point (and `modprobe wilc_spi && sleep 10 && rmmod wilc_spi`).

In order to make the first connection more stable, I do `modprobe wilc_spi && sleep 10 && rmmod wilc_spi` once before starting my AP.