linux4wilc/driver #78: p2p interface

mmader87, 2019-11-07:
I recently upgraded my kernel from 4.19 to 5.3.  After the upgrade I lost my p2p0 interface under ifconfig.  WPA Supplicant is configured for P2P and iw list shows both P2P GO and P2P client.  I was wondering if anyone has encountered this issue already?

My interfaces file is attached.
[interfaces.txt](https://github.com/linux4wilc/driver/files/3820699/interfaces.txt)

AdhamAbozaeid, 2019-11-07:
The p2p0 interface isn't added automatically now since the WILC driver started supporting dynamic interfaces, so you need to add it using 
`iw phy0 interface add p2p0 type station`
You'll need to change the phy0 to the phy on your system that get with iw list

If you have done that already, try showing all interfaces using
`ifconfig -a`
and up it using
`ifconfig p2p0 up`

mmader87, 2019-11-07:
That worked Thank you for the support!

AdhamAbozaeid, 2019-11-07:
Anytime

closed 2019-11-07 by AdhamAbozaeid