linux4wilc/driver #79: p2p GO ssid fails to broadcast

mmader87, 2019-11-08:
After I setup the wilc1000 to p2p GO autonomous, the ssid fails to broadcast.

Below are the commands I am issuing via commandline after boot.  I have attached my p2p_supplicant.conf file [p2p_supplicant.txt](https://github.com/linux4wilc/driver/files/3824802/p2p_supplicant.txt).

ifconfig wlan0 up
iw phy0 interface add p2p0 type station
ifconfig p2p0 up
wpa_supplicant -B -Dnl80211 -ip2p0 -c /etc/p2p_supplicant.conf
wpa_cli -ip2p0 p2p_group_add persistent=0

Below is a link to the log I captured.  Everything appears to be OK but I cannot find the SSID "DIRECT-XR-mydevice" from any other wifi device.
[LogOutput.txt](https://github.com/linux4wilc/driver/files/3824825/LogOutput.txt)

Using Linux Kernel v5.3 and WILC1000 driver v15.3.2.

Thank you in advance for your assistance!

ajaykathat, 2019-11-12:
From the shared debug logs, it appears the AP is started successfully on channel 6 using p2p0 interface.
Copied the below logs from share log file(LogOutpu.txt). 
[  105.239912] WILC_SPI spi0.0 p2p0: INFO [start_ap]Starting ap
[  105.244936]  Head length= 71 Tail length= 94 channelnum[6]

**"wpa_cli -ip2p0 status"** command also shows that GO interface is up, so you should see the AP beacons on the air.

Could you please confirm which wifi device you have used to check the AP presence. You can try to use a different wifi device or phone to check the AP presence.
Also, you can use **"wpa_cli -ip2p0"** command after starting the P2P_Go to know if the probe_request(RX-PROBE-REQUEST) frames are getting received by the wpa_supplicant. This is just to confirm that frames are getting received on the 'p2p0' interface successfully.

mmader87, 2019-11-12:
Thank you for your response, below is the response to the "**wpa_cli -p2p0 status**" as requested:

> root@mydevice:~# wpa_cli -ip2p0 status
> bssid=f8:f0:05:ae:ae:83
> freq=2412
> ssid=DIRECT-XR-mydevice
> id=1
> mode=P2P GO
> pairwise_cipher=CCMP
> group_cipher=CCMP
> key_mgmt=WPA2-PSK
> wpa_state=COMPLETED
> ip_address=169.254.36.170
> p2p_device_address=f8:f0:05:ae:ae:83
> address=f8:f0:05:ae:ae:83
> uuid=f87ec44d-8c78-5987-a3b1-e750175bf94a

I'm not sure how to verify the RX-PROBE-REQUEST per your message however, the wpa_cli -p2p0 command takes me to the wpa interactive mode:

> root@exactxxxxx:~# wpa_cli -ip2p0
> wpa_cli v2.7
> Copyright (c) 2004-2018, Jouni Malinen <j@w1.fi> and contributors
> 
> This software may be distributed under the terms of the BSD license.
> See README for more details.
> 
> Interactive mode

If I am expecting to see the RX-PROBE-REQUEST message inside of the interactive mode automatically, after I can have configured the p2p device for GO Mode, then I am not seeing this.

I have verified on several devices that I am not seeing the p2p SSID.  These devices include:
1) Windows 10 Laptop
2) Windows 10 Desktop w/ 3rd party Wi-Fi dongle
3) Android 9 Motorola Moto g6
4) Ubuntu 16.04LTS Laptop

Thank you for your continued support.

ajaykathat, 2019-11-12:
> > root@exactxxxxx:~# wpa_cli -ip2p0
> > wpa_cli v2.7
> > Copyright (c) 2004-2018, Jouni Malinen [j@w1.fi](mailto:j@w1.fi) and contributors
> > This software may be distributed under the terms of the BSD license.
> > See README for more details.
> > Interactive mode
> 
> If I am expecting to see the RX-PROBE-REQUEST message inside of the interactive mode >automatically, after I can have configured the p2p device for GO Mode, then I am not seeing this.

Yes, once the P2P_Go interface is up then the interactive mode will output RX-PROBE-REQUEST debug message on the console.

I suspect the issue is happening because the firmware used in your setup is old (WILC_WIFI_FW_REL_15_2_1 Build: 11257). You have used the latest driver(15.3.2) but your firmware binary is old. 

To verify please use the latest firmware from [linux4wilc/firmware](https://github.com/linux4wilc/firmware/blob/dev/wilc1000_wifi_firmware.bin). 
You can replace the /lib/firmware/mchp/wilc1000_wifi_firmware.bin file with the latest firmware from github to check.

mmader87, 2019-11-12:
I went ahead and updated the firmware per recommendation as shown below:

> [   10.292482] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC1000
> [   10.297724] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc1000_wifi_firmware.bin
> [   10.303040] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc1000_wifi_firmware.bin
> [   10.317857] WILC_SPI spi0.0 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
> [   10.323250] wilc_wlan_firmware_downloadDownloading firmware size = 135060
> [   10.398658] wilc_wlan_firmware_download Offset = 119744
> [   10.413450] wilc_wlan_firmware_download Offset = 135060
> [   10.418835] WILC_SPI spi0.0 wlan0: INFO [wilc_firmware_download]Download Succeeded
> [   10.424462] WILC_SPI spi0.0 wlan0: INFO [wilc_start_firmware]Starting Firmware ...
> [   10.430890] WILC_SPI spi0.0 wlan0: INFO [wilc_start_firmware]Waiting for FW to get ready ...
> [   10.515539] cfg_indicate_rx: Info message received
> [   10.521194] WILC_SPI spi0.0 wlan0: INFO [wilc_start_firmware]Firmware successfully started
> [   10.536010] WILC_SPI spi0.0 wlan0: INFO [wilc_wlan_initialize]WILC Firmware Ver = WILC_WIFI_FW_REL_15_2_1 Build: 11257
> [   10.541822] WILC_SPI spi0.0 wlan0: INFO [wilc_init_fw_config]Start configuring Firmware
> 

I now see the RX-PROBE-REQUEST messages as expected.  **Unfortunately, I still cannot see the SSID from any other Wi-Fi device.**

> <3>RX-PROBE-REQUEST sa=d0:ab:d5:e4:86:2b signal=0
> <3>RX-PROBE-REQUEST sa=d0:ab:d5:e4:86:2b signal=0
> <3>RX-PROBE-REQUEST sa=9c:30:5b:b5:62:01 signal=0
> <3>RX-PROBE-REQUEST sa=9c:30:5b:b5:62:01 signal=0
> <3>RX-PROBE-REQUEST sa=9c:30:5b:b5:62:01 signal=0
> <3>RX-PROBE-REQUEST sa=9c:30:5b:b5:62:01 signal=0
> <3>WPS-ENROLLEE-SEEN 00:50:03:3d:09:9d 55dd9295-9dd3-511a-8521-adc9cb606fff 0-00000000-0 0x2108 0 0 [ ]
> <3>RX-PROBE-REQUEST sa=00:50:03:3d:09:9d signal=0
> <3>WPS-ENROLLEE-SEEN 00:50:03:3d:09:9d 55dd9295-9dd3-511a-8521-adc9cb606fff 0-00000000-0 0x2108 0 0 [ ]
> <3>RX-PROBE-REQUEST sa=00:50:03:3d:09:9d signal=0
> <3>WPS-ENROLLEE-SEEN 00:50:03:3d:09:9d 55dd9295-9dd3-511a-8521-adc9cb606fff 0-00000000-0 0x2108 0 0 [ ]
> <3>RX-PROBE-REQUEST sa=00:50:03:3d:09:9d signal=0
> <3>WPS-ENROLLEE-SEEN 00:50:03:3d:09:9d 55dd9295-9dd3-511a-8521-adc9cb606fff 0-00000000-0 0x2108 0 0 [ ]
> <3>RX-PROBE-REQUEST sa=00:50:03:3d:09:9d signal=0
> <3>RX-PROBE-REQUEST sa=7c:2a:31:78:88:22 signal=0
> <3>RX-PROBE-REQUEST sa=9c:30:5b:b5:62:01 signal=0
> <3>RX-PROBE-REQUEST sa=9c:30:5b:b5:62:01 signal=0
> <3>RX-PROBE-REQUEST sa=9c:30:5b:b5:62:01 signal=0
> <3>RX-PROBE-REQUEST sa=9c:30:5b:b5:62:01 signal=0
> <3>RX-PROBE-REQUEST sa=9c:30:5b:b5:62:01 signal=0
> <3>RX-PROBE-REQUEST sa=7c:2a:31:78:88:22 signal=0
> <3>WPS-ENROLLEE-SEEN 00:50:03:3d:09:9d 55dd9295-9dd3-511a-8521-adc9cb606fff 0-00000000-0 0x2108 0 0 [ ]
> <3>RX-PROBE-REQUEST sa=00:50:03:3d:09:9d signal=0
> <3>WPS-ENROLLEE-SEEN 00:50:03:3d:09:9d 55dd9295-9dd3-511a-8521-adc9cb606fff 0-00000000-0 0x2108 0 0 [ ]
> <3>RX-PROBE-REQUEST sa=00:50:03:3d:09:9d signal=0
> <3>RX-PROBE-REQUEST sa=d4:6a:6a:6f:41:99 signal=0
> <3>RX-PROBE-REQUEST sa=9c:30:5b:b5:82:cd signal=0
> <3>RX-PROBE-REQUEST sa=9c:30:5b:b5:82:cd signal=0
> <3>RX-PROBE-REQUEST sa=9c:30:5b:b5:82:cd signal=0
> <3>RX-PROBE-REQUEST sa=d0:ab:d5:e4:86:2b signal=0
> <3>RX-PROBE-REQUEST sa=d0:ab:d5:e4:86:2b signal=0
> <3>RX-PROBE-REQUEST sa=d0:ab:d5:e4:86:2b signal=0
> <3>RX-PROBE-REQUEST sa=dc:53:60:3d:32:0e signal=0
> <3>RX-PROBE-REQUEST sa=dc:53:60:3d:32:0e signal=0
> <3>RX-PROBE-REQUEST sa=7c:2a:31:78:88:22 signal=0
> <3>WPS-ENROLLEE-SEEN 00:50:03:3d:09:9d 55dd9295-9dd3-511a-8521-adc9cb606fff 0-00000000-0 0x2108 0 0 [ ]
> <3>RX-PROBE-REQUEST sa=00:50:03:3d:09:9d signal=0
> <3>WPS-ENROLLEE-SEEN 00:50:03:3d:09:9d 55dd9295-9dd3-511a-8521-adc9cb606fff 0-00000000-0 0x2108 0 0 [ ]
> <3>RX-PROBE-REQUEST sa=00:50:03:3d:09:9d signal=0
> <3>WPS-ENROLLEE-SEEN 00:50:03:3d:09:9d 55dd9295-9dd3-511a-8521-adc9cb606fff 0-00000000-0 0x2108 0 0 [ ]
> <3>RX-PROBE-REQUEST sa=00:50:03:3d:09:9d signal=0
> <3>WPS-ENROLLEE-SEEN 00:50:03:3d:09:9d 55dd9295-9dd3-511a-8521-adc9cb606fff 0-00000000-0 0x2108 0 0 [ ]
> <3>RX-PROBE-REQUEST sa=00:50:03:3d:09:9d signal=0
> <3>WPS-ENROLLEE-SEEN 00:50:03:3d:09:9d 55dd9295-9dd3-511a-8521-adc9cb606fff 0-00000000-0 0x2108 0 0 [ ]
> <3>RX-PROBE-REQUEST sa=00:50:03:3d:09:9d signal=0
> <3>WPS-ENROLLEE-SEEN 00:50:03:3d:09:9d 55dd9295-9dd3-511a-8521-adc9cb606fff 0-00000000-0 0x2108 0 0 [ ]
> <3>RX-PROBE-REQUEST sa=00:50:03:3d:09:9d signal=0
> <3>RX-PROBE-REQUEST sa=7c:2a:31:78:88:22 signal=0
> <3>WPS-ENROLLEE-SEEN 00:50:03:3d:09:9d 55dd9295-9dd3-511a-8521-adc9cb606fff 0-00000000-0 0x2108 0 0 [ ]
> <3>RX-PROBE-REQUEST sa=00:50:03:3d:09:9d signal=0
[... 49 lines cut ...]
> <3>RX-PROBE-REQUEST sa=00:50:03:3d:09:9d signal=0
> <3>WPS-ENROLLEE-SEEN 00:50:03:3d:09:9d 55dd9295-9dd3-511a-8521-adc9cb606fff 0-00000000-0 0x2108 0 0 [ ]
> <3>RX-PROBE-REQUEST sa=00:50:03:3d:09:9d signal=0
> <3>WPS-ENROLLEE-SEEN 00:50:03:3d:09:9d 55dd9295-9dd3-511a-8521-adc9cb606fff 0-00000000-0 0x2108 0 0 [ ]
> <3>RX-PROBE-REQUEST sa=00:50:03:3d:09:9d signal=0
> <3>RX-PROBE-REQUEST sa=d4:6a:6a:6f:41:99 signal=0
> <3>RX-PROBE-REQUEST sa=d4:6a:6a:6f:41:99 signal=0
> <3>RX-PROBE-REQUEST sa=9c:30:5b:b5:09:ff signal=0
> <3>RX-PROBE-REQUEST sa=9c:30:5b:b5:09:ff signal=0
> <3>RX-PROBE-REQUEST sa=9c:30:5b:b5:82:cd signal=0
> <3>RX-PROBE-REQUEST sa=9c:30:5b:b5:82:cd signal=0
[... cut ...]
> <3>RX-PROBE-REQUEST sa=d0:ab:d5:e4:86:2b signal=0
> <3>RX-PROBE-REQUEST sa=d0:ab:d5:e4:86:2b signal=0
> <3>RX-PROBE-REQUEST sa=d0:ab:d5:e4:86:2b signal=0
> <3>RX-PROBE-REQUEST sa=d0:ab:d5:e4:86:2b signal=0
> <3>RX-PROBE-REQUEST sa=d0:ab:d5:e4:86:2b signal=0
> <3>RX-PROBE-REQUEST sa=d0:ab:d5:e4:86:2b signal=0
> <3>RX-PROBE-REQUEST sa=d0:ab:d5:e4:86:2b signal=0
> <3>RX-PROBE-REQUEST sa=dc:53:60:3d:32:0e signal=0
> <3>RX-PROBE-REQUEST sa=7c:2a:31:78:88:22 signal=0
> <3>RX-PROBE-REQUEST sa=54:13:79:74:c0:db signal=0

ajaykathat, 2019-11-12:
Now from the logs, it looks like the AP interface is up correctly and it should have worked.

As you know, the SSID will be displayed in found network list in wifi setting but if you are viewing under wifi-direct it will display the p2p device name ie. mydevice as per your '.conf' setting.
In my setup using your .conf file, I am able to see the p2p_go device correctly.

Btw any of **sa**(source address) mac address is matching with your device mac address in above RX-PROBE-REQUEST.

If possible you can also verify available network SSID using any "wifi analyzer" android app and confirm the missing SSID? or Is it possible to capture the air logs to check the wifi traffic?

If it didn't help you can try to setup p2p connection using [wilc user guide](http://ww1.microchip.com/downloads/en/DeviceDoc/ATWILC1000-ATWILC3000-Wi-Fi-Link-Controller-Linux-User-Guide-DS70005328C.pdf)(section 8.7) to check if that works for you.

mmader87, 2019-11-12:
I finally got my SSID to broadcast.  The final key was dchpcd was also trying to run on my p2p0 interface.  I'm not sure why this daemon would cause this sort of behavior, but I figured I would let you know.

Thank you all for your support! I truly do appreciate it!

mmader87, 2019-11-12:
One last question, is it possible to have the wpa_supplicant running on both wlan0 and p2p0 or is this discouraged?

Thank you for your help!

ajaykathat, 2019-11-13:
> One last question, is it possible to have the wpa_supplicant running on both wlan0 and p2p0 or is this discouraged?

Yes, it's possible to run wpa_supplicant on both wlan0 and p2p0 interfaces to handle concurrent mode. 
For supported concurrent modes and its configuration please refer section 8.8 in [ATWILC1000-ATWILC3000-Wi-Fi-Link-Controller-Linux-User-Guide](http://ww1.microchip.com/downloads/en/DeviceDoc/ATWILC1000-ATWILC3000-Wi-Fi-Link-Controller-Linux-User-Guide-DS70005328C.pdf).

closed 2019-11-13 by AdhamAbozaeid