linux4wilc/driver #82: WILC3000 Bluetooth Flow control causing SDIO errors

BDP42, 2019-11-22:
Hi,
I am trying to get the WILC3000 Bluetooth module up and running on an iMX6-based custom platform. The WILC3000 BT Uart interface is connected to one of the internal iMX6 uarts.

I am using the following commands to bring up and configure the BT device.

```
fm5000:/ # echo BT_POWER_UP > /dev/wilc_bt

at_pwr_dev: open()
received BT_POWER_UP
param len: 1, string: BT_POWER_UP
AT PWR: bt_power_up
Device already up. request source is BT
at_pwr_dev: close()

fm5000:/ # echo BT_DOWNLOAD_FW > /dev/wilc_bt
at_pwr_dev: open()
received BT_DOWNLOAD_FW
param len: 1, string: BT_DOWNLOAD_FW

AT PWR: bt_download_fw
Bluetooth firmware: mchp/wilc3000_ble_firmware.bin
atmel wilc_bt: Direct firmware load for mchp/wilc3000_ble_firmware.bin failed with error -2
atmel wilc_bt: Falling back to user helper
Downloading BT firmware size = 58184 ...
Starting BT firmware
BT Start Succeeded
at_pwr_dev: close()

fm5000:/ # hciattach ttymxc4 any 115200 noflow
Device setup complete

fm5000:/ # hciconfig hci0 up
```
At this point the interface is up and everything looks good.

```
fm5000:/ # hciconfig
hci0:   Type: Primary  Bus: UART
        BD Address: F8:F0:05:DC:9B:B1  ACL MTU: 27:30  SCO MTU: 0:0
        UP RUNNING
        RX bytes:382 acl:0 sco:0 events:24 errors:0
        TX bytes:128 acl:0 sco:0 commands:24 errors:0
```
I enable flow control on the BT device

```
fm5000:/ # hcitool cmd 0x3f 0x0053 00 c2 01 00 01
< HCI Command: ogf 0x3f, ocf 0x00 53, plen 5
  00 C2 01 00 01
> HCI Event: 0x0e plen 4
  0F 53 FC 00
```
 I then enable flow control on the iMX6 uart and it is at this point that I get continuous SDIO errors

```
stty -F /dev/ttymxc4 crtscts

wilc_sdio mmc1:0001:1: wilc_sdio_cmd53..failed, err(-84)
wilc_sdio mmc1:0001:1: Failed cmd53 [0], bytes read...
wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_handle_isr_ext:1431] fail block rx
wilc_sdio mmc1:0001:1: wilc_sdio_cmd53..failed, err(-84)
wilc_sdio mmc1:0001:1: Failed cmd53 [0], bytes read...
wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_handle_isr_ext:1431] fail block rx
wilc_sdio mmc1:0001:1: wilc_sdio_cmd53..failed, err(-84)
wilc_sdio mmc1:0001:1: Failed cmd53 [0], bytes read...
wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_handle_isr_ext:1431] fail block rx
wilc_sdio mmc1:0001:1: wilc_sdio_cmd53..failed, err(-84)
wilc_sdio mmc1:0001:1: Failed cmd53 [0], bytes read...
wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_handle_isr_ext:1431] fail block rx
wilc_sdio mmc1:0001:1: wilc_sdio_cmd53..failed, err(-84)
wilc_sdio mmc1:0001:1: Failed cmd53 [0], bytes read...
wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_handle_isr_ext:1431] fail block rx
```

Here's a trace of the UART signals when things go wrong.

[image]

Flow control on the BT device has been enabled just after 3 seconds on the trace and the device  asserts RTS.  After about 13.5 seconds flow control on the uart is enabled and it asserts CTS and it seems that at this point the errors start.
The flow control behaviour looks OK and I can't see anything obviously wrong.
Any ideas what could be causing this?
The wlan0 interface has been brought up before all of this.

Thanks,
Bruno

AdhamAbozaeid, 2019-11-22:
Hi @BDP42 
Which WILC3000 board are you using? Is it a custom design or an MCHP board?
Any chance there can be  a short between the RTS/CTS lines and any of the SDIO lines?

I ran a quick test on my sama5d4 + WILC3000 shield board, and could ping on the wlan0 interface after enabling BLE's flow control

BDP42, 2019-11-25:
Hi @AdhamAbozaeid 
It's a custom board. I don't think there is a short but I will investigate this aspect further.

BDP42, 2019-11-26:
Hi @AdhamAbozaeid 
It turns out the problem was caused by VDDIO not correctly  supplied to the device.
This has been fixed and the sdio errors are gone and can talk to the device after enabling flow contlol on both sides.
So simply a problem with the board.

closed 2019-11-26 by BDP42

AdhamAbozaeid, 2019-11-26:
Thanks for the updates