linux4wilc/driver #84: kernel NULL pointer at PC is at debug_thread+0x144/0x2e8 [wilc_sdio]

mgkiller7, 2020-01-06:
we use wilc1000 in cpu NUC976（arm926t core）with linux 3.10.108.  wilc module work well   except  for scheduling bug at https://github.com/linux4wilc/driver/issues/83（Not yet solved）. when wilc1000 sucess to connect to a AP, we continuous ping the AP's gateway ip 192.168.50.1 . when sucessfull reach 173206 times, the ping cmd fail to ping any more. At this time the wilc driver log out "cfg_get Timed Out",  and end with "kernel NULL pointer".

**Below is the log:**
8008 bytes from 192.168.50.1: seq=42118 ttl=64 time=130.000 ms
8008 bytes from 192.168.50.1: seq=42119 ttl=64 time=1020.000 ms
8008 bytes from 192.168.50.1: seq=42120 ttl=64 time=10.000 ms
8008 bytes from 192.168.50.1: seq=42121 ttl=64 time=1030.000 ms
8008 bytes from 192.168.50.1: seq=42122 ttl=64 time=20.000 ms
8008 bytes from 192.168.50.1: seq=42123 ttl=64 time=90.000 ms
[176580.360000] wilc_sdio mmc0:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running ***
8008 bytes from 192.168.50.1: seq=42124 ttl=64 time=110.000 ms
8008 bytes from 192.168.50.1: seq=42125 ttl=64 time=120.000 ms
8008 bytes from 192.168.50.1: seq=42126 ttl=64 time=1020.000 ms
8008 bytes from 192.168.50.1: seq=42127 ttl=64 time=10.000 ms
8008 bytes from 192.168.50.1: seq=42128 ttl=64 time=170.000 ms
8008 bytes from 192.168.50.1: seq=42129 ttl=64 time=70.000 ms
[176586.360000] wilc_sdio mmc0:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running *** 
8008 bytes from 192.168.50.1: seq=42130 ttl=64 time=90.000 ms
8008 bytes from 192.168.50.1: seq=42131 ttl=64 time=200.000 ms
8008 bytes from 192.168.50.1: seq=42132 ttl=64 time=120.000 ms
8008 bytes from 192.168.50.1: seq=42133 ttl=64 time=1020.000 ms
8008 bytes from 192.168.50.1: seq=42134 ttl=64 time=10.000 ms
8008 bytes from 192.168.50.1: seq=42135 ttl=64 time=160.000 ms
[176592.360000] wilc_sdio mmc0:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running ***
8008 bytes from 192.168.50.1: seq=42136 ttl=64 time=70.000 ms
**8008 bytes from 192.168.50.1: seq=42137 ttl=64 time=90.000 ms**
[176598.360000] wilc_sdio mmc0:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running *** 
[176600.370000] cfg_get Timed Out
[176600.370000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1786] Get Timed out
[176600.390000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_get_statistics:1084] Failed to send scan parameters
[176604.360000] wilc_sdio mmc0:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[176605.370000] cfg_get Timed Out
[176605.370000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1786] Get Timed out
[176605.390000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_get_statistics:1084] Failed to send scan parameters
[176610.360000] wilc_sdio mmc0:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[176610.380000] cfg_get Timed Out
[176610.380000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1786] Get Timed out
[176610.400000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_get_statistics:1084] Failed to send scan parameters
[176615.410000] cfg_get Timed Out
[176615.410000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1786] Get Timed out
[176615.420000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_get_statistics:1084] Failed to send scan parameters
[176616.360000] wilc_sdio mmc0:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[176620.400000] cfg_get Timed Out
[176620.400000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1786] Get Timed out
[176620.410000] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_get_statistics:1084] Failed to send scan parameters
[176622.360000] wilc_sdio mmc0:0001:1 wlan0: INFO [debug_thread]*** Debug Thread Running ***
[176622.360000] wilc_sdio mmc0:0001:1 wlan0: INFO [debug_thread]<Recover>
[176622.380000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_close]Mac close
[176622.380000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_close]Deinitializing wilc
[176622.400000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing wilc  ...
[176622.400000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Disabling IRQ
[176622.440000] wilc_sdio mmc0:0001:1: wilc_sdio_disable_interrupt
[176622.450000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing Threads
[176622.470000] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[176622.470000] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[176622.490000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing IRQ
[176622.500000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing WILC Wlan
[176622.510000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_deinitialize]wilc deinitialization Done
[176622.520000] Unable to handle kernel NULL pointer dereference at virtual address 00000080
[176622.530000] pgd = c0004000
[176622.530000] [00000080] *pgd=00000000
[176622.530000] Internal error: Oops: 17 [#1] PREEMPT ARM
[176622.530000] Modules linked in: wilc_sdio(C) watchdogd(O) usb_f_rndis u_ether libcomposite
[176622.530000] CPU: 0 PID: 1647 Comm: WILC_DEBUG Tainted: G        WC O 3.10.108 #5
[176622.530000] task: c38d82c0 ti: c1fd8000 task.ti: c1fd8000
[176622.530000] PC is at debug_thread+0x144/0x2e8 [wilc_sdio]
[176622.530000] LR is at wake_up_klogd+0x54/0x7c
[176622.530000] pc : [<bf0226f0>]    lr : [<c001ceb8>]    psr: 40000013
[176622.530000] sp : c1fd9f20  ip : c1fd9cf8  fp : c1fd9f64
[176622.530000] r10: 00000002  r9 : bf0349f4  r8 : bf0345ec
[... 183 lines cut ...]
--- a/linux-3.10.108/drivers/staging/wilc/wilc_netdev.c
+++ b/linux-3.10.108/drivers/staging/wilc/wilc_netdev.c
@@ -69,9 +69,13 @@ static int debug_thread(void *arg)
                timeout = 10;
                recovery_on = 1;
                wait_for_recovery = 1;
-               for (i = 0; i < WILC_NUM_CONCURRENT_IFC; i++)
-                       wilc_mac_close(wl->vif[i]->ndev);
+                for (i = 0; i < WILC_NUM_CONCURRENT_IFC; i++) {
+                        printk("close: wl->vif[%d] == %p\n", i, wl->vif[i]);
+                        if(wl->vif[i])
+                            wilc_mac_close(wl->vif[i]->ndev);
+                }
                for (i = WILC_NUM_CONCURRENT_IFC; i > 0; i--) {
+                       if(wl->vif[i-1] == NULL) {printk("open: wl->vif[%d] == %p\n", i-1, wl->vif[i-1]); continue;}
                        while (wilc_mac_open(wl->vif[i-1]->ndev) && --timeout)
                                msleep(100);

After this we don't have this kernel NULL pointer Oops any more, any the driver go on with other errors, we think the wilc driver can't recover itself from wilc_sdio_cmd52-failed, and shell must be hang if we type “ifconfig wlan0” or  type “ifconfig wlan0 down”  or type “rmmod wilc-sdio”, the only way we can do is to reboot the whole board.

after we re-power-on or reset the board, wilc work well again.

Our questions:
1、This kernel NULL pointer Oops is wilc driver bug？
2、After we modify debug_thread(void *arg) in wilc_netdev.c to avoid kernel NULL pointer Oops，but wilc_sdio_cmd52-failed remain，how can we fix this issue?

mgkiller7, 2020-01-10:
Finally i found out the reason why hang while type "rmmod wilc-sdio" after wilc_sdio_cmd52-failed . it is realate to a infinish loop"while (wait_for_recovery)" in wilc_send_config_pkt(), see my modify below:

int wilc_send_config_pkt(struct wilc_vif *vif, u8 mode, struct wid *wids,
                         u32 count)
{
        int i;
        int ret = 0;
        u32 drv = wilc_get_vif_idx(vif);
++        int cnt = 0;

        if (wait_for_recovery) {
                PRINT_INFO(vif->ndev, CORECONFIG_DBG,
                           "Host interface is suspended\n");
                while (wait_for_recovery) {
                        msleep(300);
                        cnt++;
++                        if(cnt > 10) {
++                                printk("[MCB] wilc_send_config_pkt wait_for_recovery timeout(3s)!!!!\n");
++                                ret = -ETIMEDOUT;
++                                return ret;
                        }
                }
...
}

But successfully rmmod wilc-sdio, the console shell frozen and can't input any cmd,  but can log out some extern HW irq relate printk. So i think the kernel is soft lockup after rmmod wilc-sdio.

Anyone have a advise？

ajaykathat, 2020-01-10:
@mgkiller7,

I think you are not using the latest driver code and in the recent release, the changes were done to handle a crash issue in debug recovery thread. It should help with the scenario you have mentioned in the 1st point(Long time ping issue). 
Also, the "scheduling while atomic" warning should not be observed in the latest release.
The latest release is shared at [wilc_linux_15_3](https://github.com/linux4wilc/driver/releases/tag/wilc_linux_15_3).
I suggest verifying the behavior with the latest driver and firmware release. 

A similar crash is observed in 2 ways but they are different issues.

1. **Long time ping with packet size:**
In this procedure, It's a temporary failure in SDIO CMD52 and recovery can result in success if the kernel Null pointer crash is avoided. Because during the reinitialized, the chance of SDIO CMD's success is more as its temporary failure. Please verify this scenario with the latest release or with the Kernel NULL pointer Oops fixes.
 

2. **ESD Test to wilc module:** 
There is a continuous SDIO CMD52 failure scenario and even if the recovery code doesn't crash the driver will fail to reinitialize because of continuous CMD52 failure. For recovery code success, the CMD52 should go thorough during re-initialize otherwise no use.

Regarding the shell hang issue, I suspect its happening because debug_thread and timers are not cleaned completely when there is a failure of recovery code (because of continuous CMD52 failure-point 2). 

Changes suggested in wilc_send_config_pkt() for 'while' loop termination is not required if we clear the 'wait_for_recovery' in debug_thread() function. Please try below changes and check if it helps to overcome the shell hang issue.
 ```
--- a/drivers/staging/wilc1000/wilc_netdev.c
+++ b/drivers/staging/wilc1000/wilc_netdev.c
@@ -106,6 +106,7 @@ static int debug_thread(void *arg)
 			hif_drv->hif_state = HOST_IF_IDLE;
 		}
 		recovery_on = 0;
+		wait_for_recovery = 0;
 	}
 	return 0;
 }
@@ -1234,6 +1235,12 @@ static int wilc_mac_close(struct net_device *ndev)
 		wl->open_ifcs--;
 	} else {
 		PRINT_ER(ndev, "MAC close called with no opened interfaces\n");
+		if (vif->ndev) {
+			netif_stop_queue(vif->ndev);
+
+			if (!recovery_on)
+				wilc_deinit_host_int(vif->ndev);
+		}
 		return 0;
 	}
 
@@ -1318,6 +1325,14 @@ void wilc_netdev_cleanup(struct wilc *wilc)
 		}
 	}
 
+	if (debug_running) {
+		if (&wilc->debug_thread_started)
+			complete(&wilc->debug_thread_started);
+		if (wilc->debug_thread) {
+			kthread_stop(wilc->debug_thread);
+			wilc->debug_thread = NULL;
+		}
+	}
 	wilc_wfi_deinit_mon_interface(wilc, false);
 
 	flush_workqueue(wilc->hif_workqueue);

```
Please note these changes may help to just remove the module successfully but to make wilc module work its must to have the success of SDIO CMD.

mgkiller7, 2020-01-15:
Thank you very much, ajaykathat
      I will have a try your advise.
      By the way, i really want to known why the master branch @ https://github.com/linux4wilc/driver had not updated sync to wilc_linux_15_3 you mentioned above. As i usually only git pull the master branch, and even can't notice the update of other Tags, eg. wilc_linux_15_3.
[image]
[image]

mgkiller7, 2020-01-15:
Hi， ajaykathat
     The "scheduling while atomic" warning also is observed with the latest release wilc_linux_15_3 in linux-3.10.108.

。。。
[  136.480000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Firmware Ver = WILC_WIFI_FW_REL_15_3 Build: 11606
。。。

wlan0: Trying to associa[  144.940000] BUG: scheduling while atomic: wpa_supplicant/1064/0x00000202
te with 04:9[  144.950000] Modules linked in:2:26:c7:3f:a wilc_sdio(C) watchdogd(O) usb_f_rndis u_ether libcomposite
8 (SSID='ASU[  144.960000] CPU: 0 PID: 1064 Comm: wpa_supplicant Tainted: G         C O 3.10.108 #8
S_A8_2G' fre[  144.970000] Backtrace: 
[  144.970000] [<c001128c>] (dump_backtrace+0x0/0x104) from [<c0011498>] (show_stack+0x18/0x1c)
[  144.980000]  r7:c0aef6d8
                            r6:c1ebe000 r5:c0aef6d8 r4:00000000
[  144.990000] [<c0011480>] (show_stack+0x0/0x1c) from [<c03631f8>] (dump_stack+0x20/0x28)
[  145.000000] [<c03631d8>] (dump_stack+0x0/0x28) from [<c003ffc0>] (__schedule_bug+0x48/0x60)
[  145.010000] [<c003ff78>] (__schedule_bug+0x0/0x60) from [<c0365100>] (__schedule+0x60/0x3d0)
[  145.010000]  r5:c0aef6d8 r4:c3877340
[  145.020000] [<c03650a0>] (__schedule+0x0/0x3d0) from [<c03654f8>] (schedule+0x88/0x8c)
[  145.030000] [<c0365470>] (schedule+0x0/0x8c) from [<c03641bc>] (schedule_timeout+0x14c/0x170)
[  145.030000] [<c0364070>] (schedule_timeout+0x0/0x170) from [<c03656f4>] (wait_for_common+0x100/0x1a4)
[  145.040000] [<c03655f4>] (wait_for_common+0x0/0x1a4) from [<c03657c8>] (wait_for_completion_timeout+0x14/0x18)
[  145.050000]  r9:c1ebfdb4 r8:00000008 r7:c1d6d3cc r6:c39323c0 r5:00000008
r4:c1d6d2c0
[  145.060000] [<c03657b4>] (wait_for_completion_timeout+0x0/0x18) from [<bf02962c>] (cfg_set+0x100/0x144 [wilc_sdio])
[  145.070000] [<bf02952c>] (cfg_set+0x0/0x144 [wilc_sdio]) from [<bf02997c>] (wilc_send_config_pkt+0x210/0x280 [wilc_sdio])
[  145.080000]  r9:bf034794 r8:00000001 r7:00000001 r6:c39323c0 r5:00000000
r4:c1ebfdb8
[  145.090000] [<bf02976c>] (wilc_send_config_pkt+0x0/0x280 [wilc_sdio]) from [<bf026e2c>] (wilc_frame_register+0x94/0x114 [wilc_sdio])
[  145.100000] [<bf026d98>] (wilc_frame_register+0x0/0x114 [wilc_sdio]) from [<bf01eaf8>] (wilc_mgmt_frame_register+0x90/0xa0 [wilc_sdio])
[  145.110000]  r6:c39333a8 r5:c1d6d140 r4:c1fac8e0
[  145.110000] [<bf01ea68>] (wilc_mgmt_frame_register+0x0/0xa0 [wilc_sdio]) from [<c032eab4>] (cfg80211_mlme_unregister_socket+0xc4/0xf0)
[  145.120000] [<c032e9f0>] (cfg80211_mlme_unregister_socket+0x0/0xf0) from [<c0329d78>] (nl80211_netlink_notify+0xf4/0x10c)
[  145.130000] [<c0329c84>] (nl80211_netlink_notify+0x0/0x10c) from [<c003d890>] (notifier_call_chain+0x4c/0x70)
[  145.140000] [<c003d844>] (notifier_call_chain+0x0/0x70) from [<c003da5c>] (__atomic_notifier_call_chain+0x40/0x54)
[  145.150000]  r9:c35b6de0 r8:00000008 r7:ffffffff r6:c1ebfecc r5:00000001
r4:c0b309dc
[  145.160000] [<c003da1c>] (__atomic_notifier_call_chain+0x0/0x54) from [<c003da90>] (atomic_notifier_call_chain+0x20/0x28)
[  145.170000]  r7:c35b6de0 r6:c35b6dc0 r5:00000000 r4:c1e80400
[  145.180000] [<c003da70>] (atomic_notifier_call_chain+0x0/0x28) from [<c0273980>] (netlink_release+0x148/0x284)
[  145.190000] [<c0273838>] (netlink_release+0x0/0x284) from [<c0242070>] (sock_release+0x28/0xa8)
[  145.200000]  r6:c3825470 r5:00000000 r4:c35b6dc0
[  145.200000] [<c0242048>] (sock_release+0x0/0xa8) from [<c0242104>] (sock_close+0x14/0x1c)
[  145.210000]  r5:c35ba660 r4:c1f14500
[  145.210000] [<c02420f0>] (sock_close+0x0/0x1c) from [<c008fc48>] (__fput+0xfc/0x224)
[  145.220000] [<c008fb4c>] (__fput+0x0/0x224) from [<c008fdb8>] (____fput+0x10/0x14)
[  145.230000] [<c008fda8>] (____fput+0x0/0x14) from [<c00361a4>] (task_work_run+0x90/0xac)
[  145.240000] [<c0036114>] (task_work_run+0x0/0xac) from [<c0010f70>] (do_work_pending+0xbc/0xc0)
[  145.250000]  r7:00000006 r6:c1ebffb0 r5:c000e308 r4:c1ebe000
[  145.250000] [<c0010eb4>] (do_work_pending+0x0/0xc0) from [<c000e1a4>] (work_pending+0xc/0x20)
[  145.260000]  r7:00000006 r6:00000002 r5:000cde70 r4:000cee28
。。。

ajaykathat, 2020-01-16:
Hi @mgkiller7,
Thanks for the verification. 
Regarding the failure of 'rmmod' under stress testing issue, I don't think it's related to the suggested fixes.
Please share the script to have an idea about the complete test procedure. For the overnight testing did you also enable the failure of CMD52 or CMD52 was getting successful.?

Regarding "scheduling while atomic" warning I don't see with Linux kernel 4.9.36.  Let's handle that issue as part of [Issue#50](https://github.com/linux4wilc/driver/issues/50)

mgkiller7, 2020-01-16:
Hi, ajaykathat
     Thanks for your reply.
     I can‘t observe any CMD52 error in the whole testing.
     My stress testing script is:
~ # cat /nvm/wilc_scan.sh 
while true
do
                # install wilc driver
                insmod /lib/modules/3.10.108/wilc-sdio.ko
                sleep 5
                
                # try to connect a non-existent wifi ap
                echo "ctrl_interface=/var/run/wpa_supplicant" > /tmp/wilc_wpa_supplicant.conf
                echo "update_config=1" >> /tmp/wilc_wpa_supplicant.conf
                wpa_supplicant -iwlan0 -Dnl80211 -c /tmp/wilc_wpa_supplicant.conf &
                sleep 5
                wpa_cli scan
                sleep 3
                wpa_cli scan_results
                wpa_cli -p/var/run/wpa_supplicant ap_scan 1
                wpa_cli -p/var/run/wpa_supplicant add_network
                #wpa_cli -p/var/run/wpa_supplicant set_network 0 ssid '"ASUS_A8_2G"'
                wpa_cli -p/var/run/wpa_supplicant set_network 0 ssid '"None-Exist-AP"'
                wpa_cli -p/var/run/wpa_supplicant set_network 0 key_mgmt WPA-PSK
                wpa_cli -p/var/run/wpa_supplicant set_network 0 psk '"12345678"'
                wpa_cli -p/var/run/wpa_supplicant select_network 0
                sleep 30
                #udhcpc -i wlan0 -q

                # down wlan0
                ifconfig wlan0 down

                # uninstall wilc driver
                rmmod wilc-sdio
                
                # sleep 
                sleep 2
done

=================================================================

 I found that some logs are missing when doing ifconfig wlan0 down before rmmod wilc-sdio failure.
the normal ifconfig wlan0 down is :
[12175.270000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_close]Mac close
[12175.290000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_close]Deinitializing wilc
[12175.290000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing wilc ...
[12175.310000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Disabling IRQ
[12175.310000] wilc_sdio mmc0:0001:1: wilc_sdio_disable_interrupt
**[12175.340000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing Threads
[12175.340000] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[12175.370000] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinit debug Thread
[12175.390000] Exit debug thread
[12175.390000] wilc_sdio mmc0:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[12175.430000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing IRQ
[12175.450000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing WILC Wlan
[12175.470000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_deinitialize]wilc deinitialization Done
[12175.480000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[12175.490000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[12175.500000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[12175.510000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[12175.520000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[12175.530000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[12175.540000] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed**

The bold log above is the missing log of doing the last ifconfig wlan0 down.
I can see the debug thread didn't exit yet  that time, this means  ifconfig wlan0 down is not success complete. So i think this uncomplete ifconfig wlan0 down cause rmmod wilc-sdio hang(mybe some locks didn't release， and ifconfig wlan0 down request it).

 i think static there must be dead loop in wilc_sdio_disable_interrupt() in wilc/wilc_sdio.c
void wilc_sdio_disable_interrupt(struct wilc *dev)
{
struct sdio_func *func = container_of(dev->dev, struct sdio_func, dev);
int ret;

    dev_info(&func->dev, "%s\n", __func__);

    if (sdio_intr_lock  == WILC_SDIO_HOST_IRQ_TAKEN)
            **wait_event_interruptible(sdio_intr_waitqueue,
                               sdio_intr_lock == WILC_SDIO_HOST_NO_TAKEN);**
    sdio_intr_lock  = WILC_SDIO_HOST_DIS_TAKEN;

    sdio_claim_host(func);
    ret = sdio_release_irq(func);
    if (ret < 0)
            dev_err(&func->dev, "can't release sdio_irq, err(%d)\n", ret);
    sdio_release_host(func);
    sdio_intr_lock  = WILC_SDIO_HOST_NO_TAKEN;
}

Mateusz-Gwara, 2020-01-16:
@mgkiller7 
during the last two years i noticed over and over again that repeated modprobe & rmmod od doing so while on high cpu load, causes the driver (or the whole system) to hang.

To give you an idea of how we handle this see:

```perl

my $driverstatus = `lsmod | grep wilc | grep -v -e "cfg80211" | awk '{print \$3}'`;
chomp($driverstatus);
TRACE("driverstatus: $driverstatus");

if ( $driverstatus eq "-1" ) {
    print("WILC driver hangs");
```

I've writte a bit about it in https://github.com/linux4wilc/firmware/issues/7 

Your issue is rather a side effect for us because reloading the driver is what we do in order to be able to operate the WILC3000 as an AP at all.

Here as well we can force the whole thing to crash just by doing a
```bash
ping -s 65000
```
instead of 
```bash
ping
```

mgkiller7, 2020-01-17:
Hi，Mateusz-Gwara
     Thank you for your sharing.
     Your skill is good to determine whether the rmmod wilc-sdio in hang，after this hang issue happn then i can reboot my whole board to recovery.
    But it is not a fundamental solution to this issue.

Mateusz-Gwara, 2020-01-17:
> But it is not a fundamental solution to this issue.
^^this is definitely true

Unfortunately the salesforce support practically stopped working with us, delaying tickets until they expire.

If you find a way to avoid a hanging module which forces the system to be rebooted, I would be very thankful for any information.

ajaykathat, 2020-01-17:
I don't see hang issue with "ping -s 65500" command. I am using SAMA5D4 Xplained + wilc3000 SDIO and using the latest firmware and driver. I have also tested with WILC3000 SPI and didn't observe the immediate hang issue.

@mgkiller7 
If you suspect the hang happening in wilc_sdio_disable_interrupt(), can you print the value of "sdio_intr_lock" variable to confirm? Also, keep one debug print while coming out of wilc_sdio_disable_interrupt().

1. what is the occurrence rate for this issue. Are you seeing this issue once every ~4 hrs?
2. Do you also see this issue if you do not start the wpa_supplicant( i.e scan and connect step)?

Mateusz-Gwara, 2020-01-17:
@ajaykathat The hang takes several days sometimes to occur.
Streaming a youtube video or webcam feed usually causes problems way faster.

mgkiller7, 2020-04-02:
Hi,  Mateusz-Gwara, ajaykathat
      Since update to the latest wilc driver&firmware, this issue never seen again.
Best regards,
mgkiller7

Mateusz-Gwara, 2020-04-03:
@mgkiller7 
Did you use the latest dev or master branch driver & fw???

Regards

Mateusz Gwara