linux4wilc/driver #85: WILC3000 Bluetooth testing issues using HCI commands

BDP42, 2020-02-05:
Hi,

I'm testing some basic Bluetooth functionality using the WILC3000 on our custom board.
I'm using the HCI command line tools to test basic connectivity between the WILC3000 and other Bluetooth devices and I'm having a few issues. 

I setup the BT interface and all looks good. 

```
fm5000:/ # hciconfig hci0
hci0:   Type: Primary  Bus: UART
        BD Address: F8:F0:05:DC:9D:5F  ACL MTU: 27:30  SCO MTU: 0:0
        UP RUNNING
        RX bytes:382 acl:0 sco:0 events:24 errors:0
        TX bytes:128 acl:0 sco:0 commands:24 errors:0

```
If I do hcitool lescan then I get a list of devices as expected.
However it does not see all devices. If I use my own phone then the WILC3000 does not see the phone and the phone does not see the WILC3000 device.

For a device it does see if I try to get the name then I get nothing

```
|fm5000:/ # hcitool name CD:CD:D2:E7:31:60
fm5000:/ #
```

if I try and get some information from the same remote device (one that appears in the scan) I get this

 ```
fm5000:/ # hcitool info CD:CD:D2:E7:31:60
Requesting information ...
Can't create connection: I/O error
```

Also if I do a hciconfig hci0 -a then I get the following even though I have set the name using hciconfig command.

```
fm5000:/ # hciconfig hci0 name bruno_test
fm5000:/ #
fm5000:/ #
fm5000:/ # hciconfig hci0  -a
hci0:   Type: Primary  Bus: UART
        BD Address: F8:F0:05:DC:9D:5F  ACL MTU: 27:30  SCO MTU: 0:0
        UP RUNNING
        RX bytes:61238 acl:0 sco:0 events:1394 errors:0
        TX bytes:452 acl:0 sco:0 commands:32 errors:0
        Features: 0x00 0x00 0x00 0x00 0x60 0x00 0x00 0x00
        Packet type: DM1 DH1 HV1
        Link policy:
        Link mode: SLAVE ACCEPT
Can't read local name on hci0: I/O error (5)
```

I'm trying to do some basics connectivity checks with other devices but it does not appear to be behaving as expected. 

- It does not see all devices as expected.
- Basic operations on device it does see do not seem to work.

Perhaps it is something I am doing, 
Your help would be appreciated.

Thanks,
Bruno

msalvinik, 2020-04-23:
Hi,

I have exactly the same problems reported by @BDP42 in the previous post.

Is there anyone who has these problems, or could help to resolve them? @BDP42 , do you still have these problems?

Thanks in advance, regards.
Mauro

BDP42, 2020-04-23:
@msalvinik  This particular issue was never resolved. At the time I was looking to verify the hardware design and that we could talk to the device correctly at the hardware level. What I did was enough to achieve this. 
However we have not yet used it as a fully functional Bluetooth device. 
This device has been used by other users and Bluetooth does work but it is interesting/worrying that you are seeing the same issue. 
It would be good/reassuring to get some kind of response from Microchip about this.

Bruno

jbaker0428, 2020-05-21:
I'm having similar issues. I have the WILC3000 on a custom Raspberry Pi Compute Module carrier board. It sees a bunch of mystery devices (not mine) when scanning with bluetoothctl, but it does not see my own peripherals that I'm trying to pair with (wireless headphones, keyboards, etc). 

What I **was** able to do is to make the WILC3000 discoverable. I was able to see the WILC3000 from a Pi 4 and pair in that direction. 

Startup steps:
`echo BT_POWER_UP > /dev/wilc_bt`
`echo BT_DOWNLOAD_FW > /dev/wilc_bt`
`hciattach ttyAMA0 any 115200 noflow`

`hciconfig`
> hci0:   Type: Primary  Bus: UART
        BD Address: F8:F0:05:74:00:3F  ACL MTU: 27:30  SCO MTU: 0:0
        UP RUNNING
        RX bytes:51032 acl:29 sco:0 events:2415 errors:0
        TX bytes:6165 acl:30 sco:0 commands:642 errors:0

In bluetoothctl:
>[bluetooth]# show
Controller F8:F0:05:74:00:3F (public)
        Name: raspberrypi
        Alias: raspberrypi
        Class: 0x00000000
        Powered: yes
        Discoverable: no
        Pairable: yes
        UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
        UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0532
        Discovering: yes

`systemctl status bluetooth` (note that I can't seem to set the local name):
> ● bluetooth.service - Bluetooth service
   Loaded: loaded (/lib/systemd/system/bluetooth.service; disabled; vendor preset: enabled)
   Active: active (running) since Thu 2020-05-21 09:36:24 UTC; 44min ago
     Docs: man:bluetoothd(8)
 Main PID: 848 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 2077)
   Memory: 1.0M
   CGroup: /system.slice/bluetooth.service
           └─848 /usr/lib/bluetooth/bluetoothd
May 21 09:36:24 raspberrypi bluetoothd[848]: sap-server: Operation not permitted (1)
May 21 09:36:24 raspberrypi bluetoothd[848]: Endpoint registered: sender=:1.29 path=/org/bluez/hci0/A2DP/SBC/Source/1
May 21 09:36:24 raspberrypi bluetoothd[848]: Endpoint registered: sender=:1.29 path=/org/bluez/hci0/A2DP/SBC/Source/2
May 21 09:36:24 raspberrypi bluetoothd[848]: Failed to set local name: Failed (0x03)
May 21 09:36:25 raspberrypi bluetoothd[848]: Failed to set local name: Failed (0x03)

`rfkill list`
> 1: phy1: Wireless LAN
        Soft blocked: no
        Hard blocked: no
2: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no

ajaykathat, 2020-05-21:
Hi @jbaker0428,
 
It's very common to get different results for BT scan and totally depends on the availability of devices in the surrounding.  The best way to confirm would be checking the BT snipper logs or maybe other device scan results to compare.

Please note WILC3000 doesn't support BT classic and only BLE mode is supported. 
[Microchip BT Android APP](https://play.google.com/store/apps/details?id=com.microchip.bluetooth.data&hl=en_IN) can be used to test on android phone.

Below are the steps for the BLE connection which are also mentioned in the wilc3000 User guide.

**Scan mode**
------------------
```
echo BT_POWER_UP > /dev/wilc_bt
echo BT_DOWNLOAD_FW > /dev/wilc_bt
hciattach ttyS1 any 115200 noflow
hciconfig hci0 up
ln -svf /usr/libexec/bluetooth/bluetoothd /usr/sbin
bluetoothd -p time -n &
bluetoothctl

//to scan 
[bluetooth]# scan on

//to list the found devices
[bluetooth]# device

//to get info
[bluetooth]# info <device mac-address>

//to connect
[bluetooth]# connect <device mac-address>
```

**Advertise mode**
----------------------
```
echo BT_POWER_UP > /dev/wilc_bt
echo BT_DOWNLOAD_FW > /dev/wilc_bt
hciattach ttyS1 any 115200 noflow 
ln -svf /usr/libexec/bluetooth/bluetoothd /usr/sbin
bluetoothd -n & 
hciconfig hci0 up 
hciconfig hci0 leadv
//afterwards connection can be initiated from phone side
```
Please also attach the complete logs with these steps becasue it might help to confirm if the same issue is observed in other setup.

jbaker0428, 2020-05-26:
**Advertising**

Advertising and connecting from the phone app works as expected with my Pixel 3.

> pi@raspberrypi:~ $ sudo bluetoothd -n &
[1] 849
pi@raspberrypi:~ $ bluetoothd[854]: Bluetooth daemon 5.50
bluetoothd[854]: Bluetooth management interface 1.14 initialized
bluetoothd[854]: Sap driver initialization failed.
bluetoothd[854]: sap-server: Operation not permitted (1)
bluetoothd[854]: Failed to set local name: Failed (0x03)
bluetoothd[854]: Failed to set local name: Failed (0x03)
pi@raspberrypi:~ $ bluetoothd[854]: Failed to set local name: Failed (0x03)
bluetoothd[854]: No cache for 52:FB:FE:A6:95:CE

> pi@raspberrypi:~ $ sudo bluetoothctl
Agent registered
[Pixel 3]# show
Controller F8:F0:05:74:00:29 (public)
        Name: raspberrypi
        Alias: swimclips
        Class: 0x00000000
        Powered: yes
        Discoverable: no
        Pairable: yes
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
        UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0532
        Discovering: no

> [Pixel 3]# devices
Device 52:FB:FE:A6:95:CE Pixel 3
Device 7C:D1:C3:24:31:F6 7C-D1-C3-24-31-F6
Device 08:66:98:E3:3B:05 08-66-98-E3-3B-05
Device A4:C1:38:1D:9D:3C ihoment_H6113_9D3C

[image]
**Scanning**

I killed bluetoothd from systemctl and restarted it with your method from the user guide for scanning.

This is all I got for logs out of bluetoothd when set up for scanning. No idea what those "Ignoring" messages are about. 

> pi@raspberrypi:/ $ sudo bluetoothd -p time -n &
[1] 989
pi@raspberrypi:/ $ bluetoothd[994]: Bluetooth daemon 5.50
bluetoothd[994]: Ignoring (cli) hostname
bluetoothd[994]: Ignoring (cli) wiimote
bluetoothd[994]: Ignoring (cli) autopair
bluetoothd[994]: Ignoring (cli) policy
bluetoothd[994]: Ignoring (cli) neard
bluetoothd[994]: Ignoring (cli) sap
bluetoothd[994]: Ignoring (cli) a2dp
bluetoothd[994]: Ignoring (cli) avrcp
bluetoothd[994]: Ignoring (cli) network
bluetoothd[994]: Ignoring (cli) input
bluetoothd[994]: Ignoring (cli) hog
bluetoothd[994]: Ignoring (cli) health
bluetoothd[994]: Ignoring (cli) gap
bluetoothd[994]: Ignoring (cli) scanparam
bluetoothd[994]: Ignoring (cli) deviceinfo
bluetoothd[994]: Ignoring (cli) midi
bluetoothd[994]: Ignoring (cli) battery
bluetoothd[994]: Ignoring (cli) sixaxis
bluetoothd[994]: Bluetooth management interface 1.14 initialized
bluetoothd[994]: Pair device timed out for hci0

Looks like there's less UUID info, not sure if that means anything.

> [bluetooth]# show
Controller F8:F0:05:74:00:29 (public)
        Name: BlueZ 5.50
        Alias: BlueZ 5.50
        Class: 0x00000000
        Powered: yes
        Discoverable: no
        Pairable: yes
        UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
        UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
        Modalias: usb:v1D6Bp0246d0532
        Discovering: no

I do see a lot of devices, but I don't see the Pixel 3 (52:FB:FE:A6:95:CE from the advertising test). Allegedly, it should be discoverable as long as I leave the Bluetooth settings menu open. Is there something I can do in the Microchip app for this?

> [bluetooth]# devices
Device 7C:D1:C3:24:31:F6 7C-D1-C3-24-31-F6
Device 08:66:98:E3:3B:05 08-66-98-E3-3B-05
Device 47:B6:4E:2D:2D:8F 47-B6-4E-2D-2D-8F
Device 5C:97:69:57:17:18 5C-97-69-57-17-18
Device 1E:28:3A:EE:5B:70 1E-28-3A-EE-5B-70
Device 6B:48:28:9E:CA:B6 6B-48-28-9E-CA-B6
Device 58:A4:80:35:8B:F1 58-A4-80-35-8B-F1
Device 4E:AD:FA:33:0B:AB 4E-AD-FA-33-0B-AB
Device 6F:6F:88:3B:8A:79 6F-6F-88-3B-8A-79
Device 56:D7:3C:DB:A8:35 56-D7-3C-DB-A8-35
Device 7B:BC:C5:9B:EB:70 7B-BC-C5-9B-EB-70
Device 78:77:27:ED:FF:A0 78-77-27-ED-FF-A0
Device 65:F1:C5:D2:7E:BD 65-F1-C5-D2-7E-BD

Tried to connect/pair to a few different mystery devices, resulting in the following errors:
> [bluetooth]# pair 65:F1:C5:D2:7E:BD
Failed to pair: org.bluez.Error.ConnectionAttemptFailed

> [bluetooth]# pair 1E:28:3A:EE:5B:70
Attempting to pair with 1E:28:3A:EE:5B:70
Failed to pair: org.bluez.Error.AuthenticationCanceled

> [bluetooth]# info  A4:C1:38:1D:9D:3C
Device A4:C1:38:1D:9D:3C (public)
        Name: ihoment_H6113_9D3C
        Alias: ihoment_H6113_9D3C
        Paired: no
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        RSSI: -97
[bluetooth]# connect  A4:C1:38:1D:9D:3C
Attempting to connect to A4:C1:38:1D:9D:3C
Failed to connect: org.bluez.Error.Failed

ajaykathat, 2020-05-26:
Good to know that the wilc3000 advertisement(as peripheral) worked for you.

The scan mode didn't find the phone because it was not configured in advertisement mode. I missed to mention earlier that the Microchip BLE data app doesn't support the advertisement mode.  So to verify this mode, you can try [nRfConnect](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en) and configure it to advertise. Afterward, the entry should be visible during the scan and allow to connect from WILC3000.