linux4wilc/driver #86: K_TXQ_TASK: blocked for more than 'x' seconds

mmader87, 2020-02-12:
On Yocto Linux v5.3.18, observing multiple instances of the following error message.  This error message happens after a period of time after boot-up.  Attached is my log, please let me know what other information to provide to assist in debugging.

[WILC1000_K_TXQ_TASK_LOG.txt](https://github.com/linux4wilc/driver/files/4193338/WILC1000_K_TXQ_TASK_LOG.txt)

Excerpt from log:
[ 2332.642010] INFO: task K_TXQ_TASK:508 blocked for more than 122 seconds.
[ 2332.643550]       Not tainted 5.3.18-yocto-standard #1
[ 2332.645038] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 2332.646569] K_TXQ_TASK      D    0   508      2 0x00000000
[ 2332.648088] Function entered at [<806244a8>] from [<80624708>]
[ 2332.649574] Function entered at [<80624708>] from [<80627e20>]
[ 2332.651023] Function entered at [<80627e20>] from [<80625414>]
[ 2332.652840] Function entered at [<80625414>] from [<7f0ec37c>]
[ 2332.654308] Function entered at [<7f0ec37c>] from [<80135f18>]
[ 2332.655766] Function entered at [<80135f18>] from [<801010d8>]
[ 2332.657221] Exception stack(0xb5e21fb0 to 0xb5e21ff8)
[ 2332.658699] 1fa0:                                     00000000 00000000 00000000 00000000
[ 2332.660245] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 2332.661753] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000

AdhamAbozaeid, 2020-02-12:
This is an INFO that task wilc_txq_task() have been idle for a while, which is expected if there's no TX packets to transmit

mmader87, 2020-02-12:
Thank you for the clarification!

ShonkyCH, 2020-02-12:
I think this is a legitimate warning. This tasks sits there constantly blocking. So your load average will sit at 2.00 with nothing else running which is annoying in itself.

WILC_DEBUG is the other one.

What is the reason for the tasks sitting in this state?

AdhamAbozaeid, 2020-02-12:
The task is idle waiting for a completion to be set.
When a packet is received from the IP stack, it's added to a TX queue and the completion is signaled, which unblocks wilc_txq_task(). It then starts processing the packet, sending it to the FW for transmission.
The queue is mainly used to avoid blocking the stack's context while the bus operations are completed, and also to give a chance to buffer enough Tx packets to be transmitted in one go, increasing the bus utilization in case of heavy traffic.

ShonkyCH, 2020-02-13:
Why does no other kernel module (that I've seen) do this? Surely this is not recommended practice.

AdhamAbozaeid, 2020-02-13:
ndo_start_xmit is called from an atomic context and hence can't wait. I already see other drivers using workqueues to do so if you are willing to compare (brcm80211 specifically).

Apart from that, I don't see a problem with deferring the work as an unrecommended practice in network driver implementations (please send reference if you think that's wrong). The current task-based implementation passed code reviews from staging and wireless subsystem and no issues regarding that was raised.

I believe the real question now should be why would the task increase the CPU load if it's idle waiting for a completion!

mmader87, 2020-02-13:
Would you like me to re-open this issue and provide you with more information to help solve this myster?

AdhamAbozaeid, 2020-02-13:
Please feel free to do so. Any contribution is appreciated.

mmader87, 2020-02-13:
This might be the only additional information I have at this time.

I am starting two instances of wpa_supplicant on boot via script:
[start-wifi.txt](https://github.com/linux4wilc/driver/files/4200349/start-wifi.txt)

wpa_supplicant.conf file for WLAN0: 
[wpa_supplicant.txt](https://github.com/linux4wilc/driver/files/4200352/wpa_supplicant.txt)

wpa_supplicant.conf file for P2P0 in GO Mode:
[p2p_supplicant.txt](https://github.com/linux4wilc/driver/files/4200356/p2p_supplicant.txt)

AdhamAbozaeid, 2020-02-13:
Thanks mmader87.
Are these are the steps you use to reproduce the INFO message in the original report?
I think we already understand why the INFO is printed, and that it's benign because the txq consumer task is expected to sleep till a tx packet is available for processing, so I don't think it should be a concern.
You can disable the INFO message by 
`echo 0 > /proc/sys/kernel/hung_task_timeout_secs`
as the message suggests

mmader87, 2020-02-13:
Correct, and yes I plan on disabling the message.  Thank you.

closed 2020-02-13 by AdhamAbozaeid