linux4wilc/driver #88: wilc300 client failing to connect in AP mode

jackmitch, 2020-02-28:
I have an issue where clients are struggling to connect to the a wilc3000 AP. I am running the latest git heads of both the driver and firmware. Please find attached a dmesg log, and hostapd configuration + test script. Is there any further debug I can enable, or any ideas for what the issue may be?

I have an Android Pixel 2 which manages to connect maybe 50% of the time, and an Intel based laptop which fails to connect 100% of the time. Any input would be appreciated.

[ap_log.txt](https://github.com/linux4wilc/driver/files/4266439/ap_log.txt)
[hostapd.txt](https://github.com/linux4wilc/driver/files/4266442/hostapd.txt)
[udhcpd.txt](https://github.com/linux4wilc/driver/files/4266445/udhcpd.txt)

```
root@wilc:~# hostapd -v       
hostapd v2.9
User space daemon for IEEE 802.11 AP management,
IEEE 802.1X/WPA/WPA2/EAP/RADIUS Authenticator
Copyright (c) 2002-2019, Jouni Malinen <j@w1.fi> and contributors
root@wilc:~# uname -a
Linux wilc 5.3.0 #1 SMP Fri Jan 17 15:23:52 UTC 2020 armv7l armv7l armv7l GNU/Linux
root@wilc:~# 
```

```
#! /usr/bin/env sh

function finish {
	echo "Cleaning up"
	killall udhcpd
	killall hostapd
}
trap finish INT

modprobe wilc-spi
/usr/sbin/hostapd hostapd.conf -P /run/hostapd.pid -B
ip addr flush dev wlan0
ip addr add 10.0.50.1/24 dev wlan0
udhcpd udhcpd.conf
iperf3 -s
```

AdhamAbozaeid, 2020-02-28:
Do you have more information about the cause of the failure? Does it happen during the authentication/Association, or while obtaining the IP address?

jackmitch, 2020-02-29:
Hi, so I have managed to get my laptop connected by switching to `dnsmasq` rather than `udhcpd`, I'm not sure what the difference is. However it's still very erratic when searching for the AP, it only maybe sees the beacon 10% of the time and then only manages to connect 25% of the time. I'm not sure what I can provide for debugging as it feels like it's in the firmware side and it's not passing information down to the driver.

Once connected the throughput is fairly steady, but I do see a lot of retries when using iperf as shown below.

```[jack@arch-matebook ~]$ iperf3 -c 172.24.1.1 -t 100
Connecting to host 172.24.1.1, port 5201
[  5] local 172.24.1.63 port 35524 connected to 172.24.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  1.22 MBytes  10.2 Mbits/sec    4   31.1 KBytes       
[  5]   1.00-2.00   sec  1.12 MBytes  9.38 Mbits/sec    4   29.7 KBytes       
[  5]   2.00-3.00   sec  1018 KBytes  8.34 Mbits/sec    1   33.9 KBytes       
[  5]   3.00-4.00   sec  1.30 MBytes  10.9 Mbits/sec    0   53.7 KBytes       
[  5]   4.00-5.00   sec  1.12 MBytes  9.38 Mbits/sec    2   36.8 KBytes       
[  5]   5.00-6.00   sec  1.12 MBytes  9.39 Mbits/sec    0   56.6 KBytes       
[  5]   6.00-7.00   sec  1.18 MBytes  9.90 Mbits/sec    2   55.1 KBytes       
[  5]   7.00-8.00   sec  1.24 MBytes  10.4 Mbits/sec    0   69.3 KBytes       
[  5]   8.00-9.00   sec  1.30 MBytes  10.9 Mbits/sec    1   60.8 KBytes       
[  5]   9.00-10.00  sec  1.12 MBytes  9.38 Mbits/sec    0   72.1 KBytes       
[  5]  10.00-11.00  sec  1.30 MBytes  10.9 Mbits/sec    1   62.2 KBytes       
[  5]  11.00-12.00  sec  1.12 MBytes  9.38 Mbits/sec    0   73.5 KBytes       
[  5]  12.00-13.00  sec  1.30 MBytes  11.0 Mbits/sec    1   65.0 KBytes       
[  5]  13.00-14.00  sec  1.12 MBytes  9.38 Mbits/sec    0   74.9 KBytes       
[  5]  14.00-15.00  sec  1.30 MBytes  10.9 Mbits/sec    0   87.7 KBytes       
[  5]  15.00-16.00  sec  1.12 MBytes  9.38 Mbits/sec    1   76.4 KBytes       
[  5]  16.00-17.00  sec  1.37 MBytes  11.5 Mbits/sec    0   86.3 KBytes       
^C[  5]  17.00-17.35  sec   445 KBytes  10.3 Mbits/sec    1   62.2 KBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-17.35  sec  20.8 MBytes  10.1 Mbits/sec   18             sender
[  5]   0.00-17.35  sec  0.00 Bytes  0.00 bits/sec                  receiver
iperf3: interrupt - the client has terminated
```

If you have any suggestions or extra debug I can enable, please let me know.

AdhamAbozaeid, 2020-03-02:
How busy is the environment you are testing in? Can you redo the test in a quite place or in shielded room?