linux4wilc/driver #90: Driver do not work in 64bit system with DMA limited to 32bits address

ruschigo, 2020-03-30:
Hi!

I had two custom platforms, one is based on ARM32 and the other with ARM64, i already tested the ATCWILC3000 in the ARM32 platform with success.

But when i try ATCWILC3000 with the ARM64, i got memory problems, when it generate the scatterlist during the CMD53, it do not take in a count the DMA restrictions of my platform. It try to do DMA with address bigger than 32bits, my hardware do not support DMA addressing bigger than 32bits.

I tried to create a buffer in "wilc_sdio_cmd53" function, where i can control the address to it with devm_kzalloc, by this way i can so far that the device works for almost a hour, but i steel get some problem after some time working, i get this; 

`Mar 30 11:52:01 localhost kernel: [ 2715.464842] [READ]temp_data = 4dcf0080 | cmd->buffer = 5a5b1da4 | *temp_data = 5a | *cmd->buffer = 5a
Mar 30 11:52:01 localhost kernel: [ 2715.722595] ksdioirqd/mmc1: page allocation failure: order:5, mode:0x40dc0(GFP_KERNEL|__GFP_COMP|__GFP_ZERO), nodemask=(null)
Mar 30 11:52:01 localhost kernel: [ 2715.722613] CPU: 3 PID: 815 Comm: ksdioirqd/mmc1 Not tainted 5.4.2 #3
Mar 30 11:52:01 localhost kernel: [ 2715.722615] Hardware name: Caninos Labrador 7 (DT)
Mar 30 11:52:01 localhost kernel: [ 2715.722619] Call trace:
Mar 30 11:52:01 localhost kernel: [ 2715.722632]  dump_backtrace+0x0/0x140
Mar 30 11:52:01 localhost kernel: [ 2715.722637]  show_stack+0x14/0x20
Mar 30 11:52:01 localhost kernel: [ 2715.722644]  dump_stack+0xb0/0xd8
Mar 30 11:52:01 localhost kernel: [ 2715.722651]  warn_alloc+0xf4/0x160
Mar 30 11:52:01 localhost kernel: [ 2715.722656]  __alloc_pages_slowpath+0x8e0/0xad0
Mar 30 11:52:01 localhost kernel: [ 2715.722661]  __alloc_pages_nodemask+0x228/0x270
Mar 30 11:52:01 localhost kernel: [ 2715.722667]  kmalloc_order+0x24/0x70
Mar 30 11:52:01 localhost kernel: [ 2715.722671]  kmalloc_order_trace+0x28/0xd0
Mar 30 11:52:01 localhost kernel: [ 2715.722676]  __kmalloc_track_caller+0x208/0x290
Mar 30 11:52:01 localhost kernel: [ 2715.722683]  devm_kmalloc+0x2c/0x98
Mar 30 11:52:01 localhost kernel: [ 2715.722712]  wilc_sdio_cmd53.isra.6+0x48/0x1c0 [wilc_sdio]
Mar 30 11:52:01 localhost kernel: [ 2715.722729]  wilc_sdio_read+0x24c/0x2b0 [wilc_sdio]
Mar 30 11:52:01 localhost kernel: [ 2715.722745]  wilc_handle_isr+0x124/0x520 [wilc_sdio]
Mar 30 11:52:01 localhost kernel: [ 2715.722760]  wilc_sdio_interrupt+0x3c/0x70 [wilc_sdio]
Mar 30 11:52:01 localhost kernel: [ 2715.722767]  process_sdio_pending_irqs+0x5c/0x1a8
Mar 30 11:52:01 localhost kernel: [ 2715.722771]  sdio_irq_thread+0x9c/0x198
Mar 30 11:52:01 localhost kernel: [ 2715.722775]  kthread+0x124/0x128
Mar 30 11:52:01 localhost kernel: [ 2715.722781]  ret_from_fork+0x10/0x1c
Mar 30 11:52:01 localhost kernel: [ 2715.722785] Mem-Info:
Mar 30 11:52:01 localhost kernel: [ 2715.722801] active_anon:147729 inactive_anon:27584 isolated_anon:27
Mar 30 11:52:01 localhost kernel: [ 2715.722801]  active_file:3972 inactive_file:3908 isolated_file:0
Mar 30 11:52:01 localhost kernel: [ 2715.722801]  unevictable:8 dirty:30 writeback:0 unstable:0
Mar 30 11:52:01 localhost kernel: [ 2715.722801]  slab_reclaimable:5362 slab_unreclaimable:237903
Mar 30 11:52:01 localhost kernel: [ 2715.722801]  mapped:23214 shmem:28262 pagetables:2149 bounce:0
Mar 30 11:52:01 localhost kernel: [ 2715.722801]  free:60273 free_pcp:131 free_cma:10652
Mar 30 11:52:01 localhost kernel: [ 2715.722812] Node 0 active_anon:590916kB inactive_anon:110336kB active_file:15888kB inactive_file:15632kB unevictable:32kB isolated(anon):108kB isolated(file):0kB mapped:92856kB dirty:120kB writeback:0kB shmem:113048kB shmem_thp: 0kB shmem_pmdmapped: 0kB anon_thp: 186368kB writeback_tmp:0kB unstable:0kB all_unreclaimable? no
Mar 30 11:52:01 localhost kernel: [ 2715.722825] DMA32 free:241092kB min:22528kB low:28160kB high:33792kB active_anon:590536kB inactive_anon:110336kB active_file:16232kB inactive_file:15756kB unevictable:32kB writepending:120kB present:2031616kB managed:1976072kB mlocked:32kB kernel_stack:6080kB pagetables:8596kB bounce:0kB free_pcp:524kB local_pcp:524kB free_cma:42608kB
Mar 30 11:52:01 localhost kernel: [ 2715.722826] lowmem_reserve[]: 0 0 0
Mar 30 11:52:01 localhost kernel: [ 2715.722832] DMA32: 13168*4kB (UME) 6437*8kB (UME) 3181*16kB (UMEC) 979*32kB (UMEHC) 191*64kB (UMHC) 0*128kB 0*256kB 1*512kB (C) 1*1024kB (C) 0*2048kB 10*4096kB (C) = 241112kB
Mar 30 11:52:01 localhost kernel: [ 2715.722861] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=1048576kB
Mar 30 11:52:01 localhost kernel: [ 2715.722864] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=32768kB
Mar 30 11:52:01 localhost kernel: [ 2715.722868] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=2048kB
Mar 30 11:52:01 localhost kernel: [ 2715.722872] Node 0 hugepages_total=0 hugepages_free=0 hugepages_surp=0 hugepages_size=64kB
Mar 30 11:52:01 localhost kernel: [ 2715.722874] 36191 total pagecache pages
Mar 30 11:52:01 localhost kernel: [ 2715.722877] 0 pages in swap cache
Mar 30 11:52:01 localhost kernel: [ 2715.722880] Swap cache stats: add 0, delete 0, find 0/0
Mar 30 11:52:01 localhost kernel: [ 2715.722883] Free swap  = 0kB
Mar 30 11:52:01 localhost kernel: [ 2715.722884] Total swap = 0kB
Mar 30 11:52:01 localhost kernel: [ 2715.722886] 507904 pages RAM
Mar 30 11:52:01 localhost kernel: [ 2715.722888] 0 pages HighMem/MovableOnly
Mar 30 11:52:01 localhost kernel: [ 2715.722890] 13886 pages reserved
Mar 30 11:52:01 localhost kernel: [ 2715.722892] 16384 pages cma reserved
Mar 30 11:52:01 localhost kernel: [ 2715.722894] 0 pages hwpoisoned
Mar 30 11:52:01 localhost kernel: [ 2715.722899] fail to alloc temp data
Mar 30 11:52:01 localhost kernel: [ 2715.722925] caninos-dma e0230000.dma-controller: Cannot do DMA to address 0x0000008000000000
Mar 30 11:52:01 localhost kernel: [ 2715.722932] caninos-dma e0230000.dma-controller: overflow 0x0000008000000000+180 of DMA mask ffffffff bus mask 0
`

my changes in the wilc_sdio_cmd53 are the following, i had add the dma restriction in probe function with; 
`dma_set_coherent_mask(wilc->dev,` `DMA_BIT_MASK(32));`

`static int wilc_sdio_cmd53(struct wilc *wilc, struct sdio_cmd53 *cmd)
{
	struct sdio_func *func = container_of(wilc->dev, struct sdio_func, dev);
	int size, ret;

	sdio_claim_host(func);

	u8 *temp_data;

	temp_data = devm_kzalloc(wilc->dev, cmd->count * cmd->block_size, GFP_KERNEL);
	if(!temp_data){
		pr_err("fail to alloc temp data");
	}

	func->num = cmd->function;
	func->cur_blksize = cmd->block_size;
	if (cmd->block_mode)
		size = cmd->count * cmd->block_size;
	else
		size = cmd->count;

	if (cmd->read_write) {  /* write */
		if (cmd->block_mode)
			memcpy(temp_data, cmd->buffer, cmd->count * cmd->block_size);
		else
			memcpy(temp_data, cmd->buffer, cmd->count);
		pr_info("[WRITE]temp_data = %x | cmd->buffer = %x | *temp_data = %x | *cmd->buffer = %x", temp_data,cmd->buffer, *temp_data,*cmd->buffer);

		ret = sdio_memcpy_toio(func, cmd->address,
				       (void *)temp_data, size);
	} else {        /* read */
		ret = sdio_memcpy_fromio(func, (void *)temp_data,
					 cmd->address,  size);
		if (cmd->block_mode)
			memcpy(cmd->buffer, temp_data, cmd->count * cmd->block_size);
		else
			memcpy(cmd->buffer, temp_data, cmd->count);
		pr_info("[READ]temp_data = %x | cmd->buffer = %x | *temp_data = %x | *cmd->buffer = %x", temp_data,cmd->buffer, *temp_data,*cmd->buffer);
	}
	sdio_release_host(func);

	if (ret)
		dev_err(&func->dev, "%s..failed, err(%d)\n", __func__,  ret);

	return ret;
}`

There are any better way to pass the DMA restriction for the ATCWIL3000's driver?

thanks!

best regards

Igor Ruschi

ruschigo, 2020-03-30:
Hi

Just to update, i had found my own mistake in my adaptation of  wilc_sdio_cmd53, i forgot to do a kfree to my pointer, i believe it solve my memory problem for while.

But my question persist, there are any better way to do that ?

Best Regards

Igor Ruschi

ruschigo, 2020-03-31:
Hi!

Just to leave a clean log, just after modprobe of wilc-sdio (with my adaptation, without it i can't even get the chipid).

[   50.023772] Registering wifi device
[   50.023784] Max scan ids= 10,Max scan IE len= 1000,Signal Type= 1,Interface Modes= 844
[   50.023952] Initializing Locks ...
[   50.025042] wifi_pm : 0
[   50.025061] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[   50.025078] wifi_pm : 1
[   50.025082] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[   50.030354] wilc_sdio mmc1:0001:1: Driver Initializing success
[   50.097161] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   50.097520] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[000000003f83f14c] wlan0
[   50.097533] WILC POWER UP
[   50.097542] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[000000003f83f14c][000000002b56d1e5]
[   50.097560] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[   50.097566] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[   50.097578] wilc_sdio mmc1:0001:1: SDIO speed: 50000000
[   50.102732] wilc_sdio mmc1:0001:1: chipid 003000d0
[   50.103285] wilc_sdio mmc1:0001:1 wlan0: INFO [init_chip]Bootrom sts = d
[   50.103424] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
[   50.103432] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
[   50.103437] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
[   50.121081] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
[   50.121242] wilc_sdio mmc1:0001:1: failed to get IRQ GPIO, load default
[   50.121321] wilc_sdio mmc1:0001:1 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 115
[   50.121543] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC3000
[   50.121553] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc3000_wifi_firmware.bin
[   50.121558] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc3000_wifi_firmware.bin
[   50.133734] wilc_sdio mmc1:0001:1: firmware: direct-loading firmware mchp/wilc3000_wifi_firmware.bin
[   50.133750] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
[   50.133756] wilc_wlan_firmware_downloadDownloading firmware size = 136788
[   50.143398] wilc_wlan_firmware_download Offset = 119860
[   50.144719] wilc_wlan_firmware_download Offset = 136788
[   50.144747] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Download Succeeded
[   50.144800] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_start_firmware]Starting Firmware ...
[   50.145401] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_start_firmware]Waiting for FW to get ready ...
[   50.242639] cfg_indicate_rx: Info message received
[   50.242740] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_start_firmware]Firmware successfully started
[   50.250847] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Firmware Ver = WILC_WIFI_FW_REL_15_3_1 Build: 11736
[   50.250859] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_fw_config]Start configuring Firmware
[   50.369733] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]Mac address: f8:f0:05:d9:2a:3e
[   50.369757] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 0.
[   50.369763] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Enable mcast filter retrive directed pkts only
[   50.370021] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 1.
[   50.370035] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 01:00:5e:00:00:01
[   50.370086] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 2.
[   50.370098] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 01:00:5e:00:00:01
[   50.370104] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 33:33:00:00:00:01
[   50.370268] wilc_sdio mmc1:0001:1 wlan0: INFO [set_power_mgmt]dev [wlan0]
[   50.370275] wilc_sdio mmc1:0001:1 wlan0: INFO [set_power_mgmt] Power save Enabled= 0 , TimeOut = -1
[   50.415240] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 2.
[   50.415260] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 01:00:5e:00:00:01
[   50.415266] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 33:33:00:00:00:01
[   50.415442] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_close]Mac close
[   50.436626] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_close]Deinitializing wilc
[   50.436637] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing wilc  ...
[   50.436642] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Disabling IRQ
[   50.436648] wilc_sdio mmc1:0001:1: wilc_sdio_disable_interrupt
[   50.436726] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing Threads
[   50.436732] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[   50.436738] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinit debug Thread
[   51.141063] Exit debug thread
[   51.141125] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[   51.141158] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing IRQ
[... cut ...]
[   57.200702] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 01:00:5e:00:00:01
[   57.200708] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 33:33:00:00:00:01
[   57.200713] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[2]: 33:33:00:00:00:fb
[   57.200719] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[3]: 33:33:ff:35:af:e7
[   57.200724] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[4]: 01:00:5e:00:00:fb
[   57.618403] NOHZ: local_softirq_pending 08

Best Regards,

Igor Ruschi

ruschigo, 2020-04-01:
After some trouble i can get the error log, after some operations, it crash, i believe it happens during large packets transmission/reception, i let the device doing pings for a hour and no error happens, but when i started downloads or uploads it always crash very fast.

Feb 14 08:12:35 localhost kernel: [   40.039842] EXT4-fs (mmcblk2p1): mounted filesystem with ordered data mode. Opts: (null)
Feb 14 08:13:06 localhost kernel: [   71.082296] Registering wifi device
Feb 14 08:13:06 localhost kernel: [   71.082308] Max scan ids= 10,Max scan IE len= 1000,Signal Type= 1,Interface Modes= 844
Feb 14 08:13:06 localhost kernel: [   71.082461] Initializing Locks ...
Feb 14 08:13:06 localhost kernel: [   71.087238] wifi_pm : 0
Feb 14 08:13:06 localhost kernel: [   71.087257] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
Feb 14 08:13:06 localhost kernel: [   71.087274] wifi_pm : 1
Feb 14 08:13:06 localhost kernel: [   71.087278] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
Feb 14 08:13:06 localhost kernel: [   71.105073] wilc_sdio mmc1:0001:1: Driver Initializing success
Feb 14 08:13:06 localhost kernel: [   71.151002] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Feb 14 08:13:06 localhost kernel: [   71.151200] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[00000000cf73d198] wlan0
Feb 14 08:13:06 localhost kernel: [   71.151204] WILC POWER UP
Feb 14 08:13:06 localhost kernel: [   71.151212] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[00000000cf73d198][00000000f8683182]
Feb 14 08:13:06 localhost kernel: [   71.151223] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
Feb 14 08:13:06 localhost kernel: [   71.151230] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
Feb 14 08:13:06 localhost kernel: [   71.151238] wilc_sdio mmc1:0001:1: SDIO speed: 50000000
Feb 14 08:13:07 localhost kernel: [   71.166000] wilc_sdio mmc1:0001:1: chipid 003000d0
Feb 14 08:13:07 localhost kernel: [   71.166451] wilc_sdio mmc1:0001:1 wlan0: INFO [init_chip]Bootrom sts = d
Feb 14 08:13:07 localhost kernel: [   71.166551] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
Feb 14 08:13:07 localhost kernel: [   71.166556] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
Feb 14 08:13:07 localhost kernel: [   71.166561] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
Feb 14 08:13:07 localhost kernel: [   71.167407] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
Feb 14 08:13:07 localhost kernel: [   71.167506] wilc_sdio mmc1:0001:1: failed to get IRQ GPIO, load default
Feb 14 08:13:07 localhost kernel: [   71.167559] wilc_sdio mmc1:0001:1 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 115
Feb 14 08:13:07 localhost kernel: [   71.167659] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC3000
Feb 14 08:13:07 localhost kernel: [   71.167664] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc3000_wifi_firmware.bin
Feb 14 08:13:07 localhost kernel: [   71.167669] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc3000_wifi_firmware.bin
Feb 14 08:13:07 localhost kernel: [   71.181980] wilc_sdio mmc1:0001:1: firmware: direct-loading firmware mchp/wilc3000_wifi_firmware.bin
Feb 14 08:13:07 localhost kernel: [   71.182007] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
Feb 14 08:13:07 localhost kernel: [   71.182021] wilc_wlan_firmware_downloadDownloading firmware size = 136788
Feb 14 08:13:07 localhost kernel: [   71.190090] wilc_wlan_firmware_download Offset = 119860
Feb 14 08:13:07 localhost kernel: [   71.191607] wilc_wlan_firmware_download Offset = 136788
Feb 14 08:13:07 localhost kernel: [   71.191626] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Download Succeeded
Feb 14 08:13:07 localhost kernel: [   71.191664] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_start_firmware]Starting Firmware ...
Feb 14 08:13:07 localhost kernel: [   71.192267] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_start_firmware]Waiting for FW to get ready ...
Feb 14 08:13:07 localhost kernel: [   71.289346] cfg_indicate_rx: Info message received
Feb 14 08:13:07 localhost kernel: [   71.289384] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_start_firmware]Firmware successfully started
Feb 14 08:13:07 localhost kernel: [   71.297217] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Firmware Ver = WILC_WIFI_FW_REL_15_3_1 Build: 11736
Feb 14 08:13:07 localhost kernel: [   71.297227] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_fw_config]Start configuring Firmware
Feb 14 08:13:07 localhost kernel: [   71.414624] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]Mac address: f8:f0:05:d9:2a:3e
Feb 14 08:13:07 localhost kernel: [   71.414650] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 0.
Feb 14 08:13:07 localhost kernel: [   71.414655] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Enable mcast filter retrive directed pkts only
Feb 14 08:13:07 localhost kernel: [   71.414862] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 1.
Feb 14 08:13:07 localhost kernel: [   71.414869] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 01:00:5e:00:00:01
[... cut ...]
Apr  1 11:06:45 localhost kernel: [  391.036752] rcu: 	(detected by 3, t=14712 jiffies, g=9457, q=5675)
Apr  1 11:06:45 localhost kernel: [  391.036759] Task dump for CPU 1:
Apr  1 11:06:45 localhost kernel: [  391.036762] swapper/1       R  running task        0     0      1 0x0000000a
Apr  1 11:06:45 localhost kernel: [  391.036770] Call trace:
Apr  1 11:06:45 localhost kernel: [  391.036781]  __switch_to+0x94/0xe8
Apr  1 11:06:45 localhost kernel: [  391.036783]            (null)

I saw that this [issue](https://github.com/linux4wilc/driver/issues/76) is very close of my...

jorgaSomDevices, 2020-04-09:
We have same issue, obtained with:
-  Boards MCIMX6ULL-EVK and ATWILC3000-SD (tested with kernels 4.9.88 and 4.19.35) with latest driver (15.3.1) and firmware (tested with 15.3.1 and 15.4 RC1).

- The results were obtained using iperf2 (2.0.12)
---------------------------------------------------------------------------------
The obtained error log is:
[  4]  0.0-10.0 sec  31.3 MBytes  26.2 Mbits/sec   0.297 ms    0/22293 (0%)
[  3] local 192.168.4.11 port 5001 connected with 192.168.4.1 port 54568
get_if_handler Invalid handle
wilc_wlan_handle_rx_buff: wilc_netdev in wilc is NULL
wilc_wlan_handle_rx_buff: Data corrupted 0, 0
wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000182
get_if_handler Invalid handle
wilc_wlan_handle_rx_buff: wilc_netdev in wilc is NULL
wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00004fea
cfg_get Timed Out
wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_send_config_pkt:1800] Get Timed out
wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_get_statistics:1106] Failed to send scan parameters
Unable to handle kernel paging request at virtual address ffff8cc4
pgd = 80004000
[ffff8cc4] *pgd=8bf5e861, *pte=00000000, *ppte=00000000
Internal error: Oops: 37 [#1] PREEMPT SMP ARM
Modules linked in: evbug wilc_sdio
CPU: 0 PID: 203 Comm: K_TXQ_TASK Not tainted 4.9.88-04802-g84dfb5dab852-dirty #115
Hardware name: Freescale i.MX6 UltraLite (Device Tree)
task: 887f0b00 task.stack: 88904000
PC is at wilc_wlan_txq_remove_from_head+0x4c/0x10c [wilc_sdio]
LR is at preempt_count_add+0xd4/0x14c
pc : [<7f00ce08>]    lr : [<8014e52c>]    psr: a00e0093
sp : 88905d40  ip : 0acbd000  fp : 00000000
r10: 88905dd4  r9 : 88905dc2  r8 : 0000000b
r7 : 200e0013  r6 : 884ae4e4  r5 : ffff8cc0  r4 : 884ae420
r3 : 884ae630  r2 : ffff8cc0  r1 : 884aed74  r0 : 200e0013
Flags: NzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment none
Control: 10c53c7d  Table: 88a7c06a  DAC: 00000051
Process K_TXQ_TASK (pid: 203, stack limit = 0x88904210)
Stack: (0x88905d40 to 0x88906000)
5d40: 88905dc0 887b6d00 88980000 0000000b 88905e04 00000060 884ae500 7f00f11c
5d60: 00000400 00000000 00000000 887404c0 7f013fa0 884aeb64 00000001 7f0185d4
5d80: 7f018600 884ae420 88905da8 884ae4cc 88905f3c 00000001 7f01a414 88980000
5da0: f0f8aea7 0d010000 01010101 00000001 00000000 00000000 00000000 00000000
5dc0: 887f0b00 8bb38a78 000029aa 00000000 b243dcee 00000006 000029aa 00000000
5de0: 8bb38a40 80e7ba40 887f0b00 8bb38a40 00000000 80160774 00000000 8bb38a40
5e00: 00000418 00000000 8bb38a40 00000000 0000000a 00000000 80f02d00 00000001
5e20: 887f0b80 0000003d 8bb38a78 80e7ba40 8bb38a40 887f0b00 88905e8c 809f8e5c
5e40: 8bb38a40 8014e1e4 80f03244 887f0e40 ffffffff 00000000 80f00000 8014a6fc
5e60: 887f0b00 00000000 80f00000 8bb38a40 887f0b00 80f06600 882398c0 00000000
5e80: 8bb35440 400e0013 80e78440 80f03244 00000001 809f8a88 02400000 8bb35440
5ea0: 80e78440 80180c74 88905ef0 88905ef0 8bb35440 88905ef0 00000001 809f8e08
5ec0: 8bb35440 80180cdc 00000000 400e0013 88905ee4 88905ef0 8bb35440 80180e98
5ee0: fffff2c8 809f801c 7fffffff 400e0013 00000200 00000000 fffff2c8 80180eb8
5f00: 887f0b00 884ae420 ffffe000 7f01a40c 884acf88 ffffff97 884ae450 884ae448
5f20: 00000001 7f005560 88740000 7f015e94 884ae538 884ae548 888f5840 00000001
5f40: 884ae420 00000000 888f5840 88904000 884ae420 7f0054c4 00000000 00000000
5f60: 00000000 80149344 00000000 00000000 887f0b00 884ae420 00000000 00000000
5f80: 88905f80 88905f80 00000000 00000000 88905f90 88905f90 888f5840 8014923c
5fa0: 00000000 00000000 00000000 80107950 00000000 00000000 00000000 00000000
5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
5fe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
[<7f00ce08>] (wilc_wlan_txq_remove_from_head [wilc_sdio]) from [<7f00f11c>] (wilc_wlan_handle_txq+0xb64/0xd80 [wilc_sdio])
[<7f00f11c>] (wilc_wlan_handle_txq [wilc_sdio]) from [<7f005560>] (wilc_txq_task+0x9c/0x248 [wilc_sdio])
[<7f005560>] (wilc_txq_task [wilc_sdio]) from [<80149344>] (kthread+0x108/0x110)
[<80149344>] (kthread) from [<80107950>] (ret_from_fork+0x14/0x24)
Code: e0234893 e5935744 e3550000 0a000016 (e5952004) 
---[ end trace 875555a42b5ff364 ]---
note: K_TXQ_TASK[203] exited with preempt_count 1
---------------------------------------------------------------------------------
**@ruschigo did you manage to find a solution?**

ruschigo, 2020-04-09:
@jorgaSomDevices 

No, not yet, i'm trying a new way, if i find any solution i will post it here.

Best Regards

Igor Ruschi

jorgaSomDevices, 2020-04-09:
@ruschigo ok, thank you.
I'm also testing, if I find something I will post it here too.
Best Regards.