linux4wilc/driver #92: WILC WiFi Driver Initializes with Success, but Fails when Enabling WiFi

dragon-123456, 2020-05-21:
Hello! I'm working with a WILC3000 module for OpenWrt. I compiled the wilc-sdio.ko for the OpenWrt Linux version 4.14.176 and the driver initialized successfully when using insmod. But when I tried to use "ifconfig wlan0 up" to bring up the WIFi, there are some I/O errors.

Here are the messages when loading the driver module.
> root@OpenWrt:/# insmod wilc-sdio.ko 
> [   22.656473] Registering wifi device
> [   22.660093] Max scan ids= 10,Max scan IE len= 1000,Signal Type= 1,Interface Modes= 844
> [   22.669102] Initializing Locks ...
> [   22.683174] wifi_pm : 0
> [   22.697492] mmc mmc0:0001: GPIO lookup for consumer reset
> [   22.727742] mmc mmc0:0001: using device tree for GPIO lookup
> [   22.740756] of_get_named_gpiod_flags: parsed 'reset-gpios' property of node '/soc/esdhc@1560000/wilc_sdio@0[0]' - )
> [   22.758102] no flags found for reset
> [   22.767660] wilc_sdio mmc0:0001:1: succesfully got gpio_reset
> [   22.781464] mmc mmc0:0001: GPIO lookup for consumer chip_en
> [   22.789822] mmc mmc0:0001: CONFIG_OF has been enabled
> [   22.801451] mmc mmc0:0001: of_node should be good
> [   22.819485] mmc mmc0:0001: using device tree for GPIO lookup
> [   22.829501] of_get_named_gpiod_flags: parsed 'chip_en-gpios' property of node '/soc/esdhc@1560000/wilc_sdio@0[0]' )
> [   22.842498] no flags found for chip_en
> [   22.846366] wilc_sdio mmc0:0001:1: succesfully got gpio_chip_en
> [   22.852514] wilc_sdio mmc0:0001:1: power is 1
> [   22.856989] wifi_pm : 1
> [   22.859487] wilc->dt_dev address: ffff800038fdd808
> [   22.864344] mmc mmc0:0001: GPIO lookup for consumer reset
> [   22.869803] mmc mmc0:0001: CONFIG_OF has been enabled
> [   22.874912] mmc mmc0:0001: of_node should be good
> [   22.879670] mmc mmc0:0001: using device tree for GPIO lookup
> [   22.885423] of_get_named_gpiod_flags: parsed 'reset-gpios' property of node '/soc/esdhc@1560000/wilc_sdio@0[0]' - )
> [   22.896734] no flags found for reset
> [   22.900359] wilc_sdio mmc0:0001:1: succesfully got gpio_reset
> [   22.906272] mmc mmc0:0001: GPIO lookup for consumer chip_en
> [   22.911903] mmc mmc0:0001: CONFIG_OF has been enabled
> [   22.917011] mmc mmc0:0001: of_node should be good
> [   22.921782] mmc mmc0:0001: using device tree for GPIO lookup
> [   22.927535] of_get_named_gpiod_flags: parsed 'chip_en-gpios' property of node '/soc/esdhc@1560000/wilc_sdio@0[0]' )
> [   22.939045] no flags found for chip_en
> [   22.942886] wilc_sdio mmc0:0001:1: succesfully got gpio_chip_en
> [   22.948981] wilc_sdio mmc0:0001:1: power is 0
> [   22.959294] wilc_sdio mmc0:0001:1: Driver Initializing success

Here are the messages when I tried to enable WiFi.

> root@OpenWrt:/# ifconfig wlan0 up
> [  375.341741] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[ffff800037ec0000] wlan0
> [  375.350334] power up request for already powered up source Wifi
> [  375.356426] Device already up. request source is Wifi
> [  375.361543] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_init_host_int]Host[ffff800037ec0000][ffff800037ec1d28]
> [  375.371287] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
> [  375.378406] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
> [  375.386224] wilc_sdio mmc0:0001:1: SDIO speed: 50000000
> [  375.391580] wilc_sdio mmc0:0001:1: wilc_sdio_cmd52..failed, err(-110)
> [  375.398107] wilc_sdio mmc0:0001:1: Fail cmd 52, enable csa...
> [  375.403937] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_wlan_initialize:905] Initializing WILC_Wlan FAILED
> [  375.413256] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_mac_open:1029] Failed to initialize wilc
> ifconfig: SIOCSIFFLAGS: I/O error

And here is my device tree for SDIO.

> wilc_sdio@0 {
> 		compatible = "microchip,wilc1000", "microchip,wilc3000";
> 		irq-gpios = <&gpio0 21 0>;
> 		reset-gpios = <&gpio0 22 0>;
> 		chip_en-gpios = <&gpio0 24 0>;
> 		reg = <0>;
> 		bus-width = <4>;
> 		status = "okay";
> 	};

**The only uncertainty I have is that for the board I'm working on, the chip_en of WILC3000 is not connecting to any line. In order for the driver to work, I gave a fake pin <&gpio0 24 0> for it to connect in the device tree. But from the schematics the chip_en should be always 1 by the pull-up register if not connected to anything. And I believe it shouldn't affect the power on sequence.**

[image]

Here are my questions:

- What causes the I/O error after the driver being initialized successfully?
- "wilc_sdio_cmd52..failed, err(-110)" indicates a TIMEOUT issue, what can be the cause of that?
- Does the chip_en have to be connected to host, or it can be fine to leave it unconnected for the power on sequence to work?

Thank you!

ajaykathat, 2020-05-21:
Hi @dragon-123456 ,

From the shared description, it looks like the first SDIO command itself was failed.  The driver module is loaded successfully.  The purpose to keep "chip_en" high is to ensure that wilc_sdio drivers probe function is called.

The IO failure could be because of the host unable to communicate with SDIO. There was a similar issue report earlier on kernel v4.14 and its possible [fix](https://github.com/linux4wilc/driver/issues/39#issuecomment-504101606) was by removing "non-removable" from wilc_sdio node's parent. Please check the board's DT and try removing the property if it helps.

dragon-123456, 2020-05-22:
Hi @ajaykathat ,

Thank you for your reply.
Although it made no difference when I removed "non-removable" from device tree, you pointed me to the right direction. I disabled the wilc_wlan_power() function in the wilc_wlan_power_on_sequence() 
as follows:

```
int wilc_wlan_power_on_sequence(struct wilc *wilc)
{
	int ret;
#if 0
	ret = wilc_wlan_power(wilc, 0);
	if (ret)
		return ret;
	ret = wilc_wlan_power(wilc, 1);
	if (ret)
		return ret;
#endif
        return 0;
}
int wilc_wlan_power_off_sequence(struct wilc *wilc)
{
	int ret;
#if 0
	ret = wilc_wlan_power(wilc, 0);
	if (ret)
		return ret;
#endif
	return 0;
}
```
And then I did a manual unbinding and binding of the SDIO device as follows:
Note that the GPIO 502 is the reset pin for the SDIO WiFi and chip_en is always 1 by pull up.
[image]

After that, I can get pass the wilc_sdio_cmd52..failed, err(-110) error I mentioned in my earlier post. However, there is a new error:

> root@OpenWrt:/sys/bus/platform/drivers/sdhci-esdhc# ifconfig wlan0 up
> [  194.261447] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[ffff800037f7c000] wlan0
> [  194.270126] WILC POWER UP
> [  194.272900] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_init_host_int]Host[ffff800037f7c000][ffff800037f7dd28]
> [  194.282671] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
> [  194.289794] wilc_sdio mmc0:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
> [  194.297612] wilc_sdio mmc0:0001:1: SDIO speed: 50000000
> [  194.302962] wilc_sdio mmc0:0001:1: Succeed cmd 52, enable csa!
> [  194.308894] wilc_sdio mmc0:0001:1: Succeed cmd 52, set func 0 block size!
> [  194.315784] wilc_sdio mmc0:0001:1: Succeed cmd 52, set IOE register!
> [  194.322222] wilc_sdio mmc0:0001:1: Succeed func 1 is ready!
> [  194.327894] wilc_sdio mmc0:0001:1: Succeed set func 1 block size!
> [  194.334084] wilc_sdio mmc0:0001:1: Succeed cmd 52, set IEN register!
> [  194.340661] wilc_sdio mmc0:0001:1: **### Unsupported chipid: 0**
> [  194.346035] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_wlan_initialize:905] Initializing WILC_Wlan FAILED
> [  194.355337] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_mac_open:1029] Failed to initialize wilc
> ifconfig: SIOCSIFFLAGS: I/O error

A returned chip ID 0 may indicate that the WILC device is still not initialized correctly.
The posts [39](https://github.com/linux4wilc/driver/issues/39) and [38](https://github.com/linux4wilc/driver/issues/38) are my references.

ajaykathat, 2020-05-26:
I think bind/unbinding the driver should have solved this. This issue could be related to WILC device not initialized properly. 

I suspect there might be a problem with SD DAT[0-3] lines because the initial commands which use for CMD52 worked but failed for CMD53(to fetch the chip-id). Please check with the host setup if DAT is configured correctly and able to handle CMD53 transaction.

You can also open a salesforce ticket by following [Submit a Case](https://microchipsupport.force.com/s/article/How-to-submit-a-case) for further assistance with debugging.

ajaykathat, 2020-06-19:
As you have confirmed that the failure happens for sdio_memcpy_fromio() issue so it might be good to check the host document to configure the SDIO node in the Device tree. Before checking host controller driver code it's better to confirm Device tree entry is configured correctly.

shaiksyedali, 2023-09-12:
Hi Guys. I am facing a similar issue as well.

This is my device tree -

```
+&usdhc1 {

+   #address-cells = <1>;

+   #size-cells = <0>;

+   pinctrl-names = "default", "state_100mhz", "state_200mhz";

+   pinctrl-0 = <&pinctrl_usdhc1>;

+   pinctrl-1 = <&pinctrl_usdhc1_100mhz>;

+   pinctrl-2 = <&pinctrl_usdhc1_200mhz>;

+   bus-width = <4>;

+        no-1-8-v;

+   pm-ignore-notify;

+   non-removable;

+   /*keep-power-in-suspend;*/

+   /* /delete-property/ vmmc-supply; */

+   vmmc-supply = <&reg_3p3v>;

+   mmc-pwrseq = <&wifi_pwrseq>;

+   status = "okay";

+

+   wilc_sdio: wilc_sdio@0{

+       compatible = "microchip,wilc3000";

+                pinctrl-names = "default";

+                pinctrl-0 = <&pinctrl_wilc_irq>;

+

+       interrupt-parent = <&gpio2>;

+       interrupts = <6 0>;

+       interrupt-names = "wilc-irq";

+

+       status = "okay";

+       reg = <0>;

+       bus-width = <4>;

+

+   };

+};
```

```
         wifi_pwrseq: wifi_pwrseq {

                 compatible = "mmc-pwrseq-wilc";

-                reset-gpios = <&gpio2 11 GPIO_ACTIVE_HIGH>;

-                powerdown-gpios = <&gpio2 10 GPIO_ACTIVE_HIGH>;

+                reset-gpios = <&gpio5 1 GPIO_ACTIVE_HIGH>;

+                powerdown-gpios = <&gpio5 0 GPIO_ACTIVE_HIGH>;

                 pinctrl-0 = <&pinctrl_wilc_pwrseq>;

                 pinctrl-names = "default";

                 status = "okay";

@@ -170,20 +170,33 @@

        >;

    };
```

This is the error I am getting,

[  609.211886] wilc_sdio mmc0:0001:1: wilc_sdio_cmd52..failed, err(-110)
[  609.222436] wilc_sdio mmc0:0001:1: Failed cmd52, set 0x10c data...
[  609.235150] wilc_sdio mmc0:0001:1: wilc_sdio_cmd52..failed, err(-110)
[  609.242865] wilc_sdio mmc0:0001:1: Failed cmd52, set 0x10c data...
[  609.255791] wilc_sdio mmc0:0001:1 wlan0: ERR [init_chip:1822] fail read reg 0x1118
[  609.269142] wilc_sdio mmc0:0001:1: wilc_sdio_cmd52..failed, err(-110)
[  609.277290] wilc_sdio mmc0:0001:1: Failed cmd 52, read reg (000000f0) ...
[  609.294456] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_wlan_initialize:836] Initializing WILC_Wlan FAILED
[  609.312282] wilc_sdio mmc0:0001:1 wlan0: ERR [wilc_mac_open:955] Failed to initialize wilc

Any pointers on this? @dragon-123456  @ajaykathat