linux4wilc/driver #94: Raspberry pi compute modprobe fails

mintu19, 2020-06-19:
I have installed drivers on Raspbian and trying to use the driver on Pi Compute module.
I have attached enable to direct 3.3V rail and reset to gpio 0. IRQ is left as it is.

I get this error on modprobe...

```
[  870.325026] wilc_sdio: module is from the staging directory, the quality is unknown, you have been warned.
[  870.335990] Registering wifi device
[  870.336006] Max scan ids= 10,Max scan IE len= 1000,Signal Type= 1,Interface Modes= 844
[  870.336504] Initializing Locks ...
[  870.343836] wifi_pm : 0
[  870.343875] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[  870.343898] wilc_sdio mmc1:0001:1: failed to get default Reset GPIO
[  870.343936] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_send_config_pkt:1806] Set Timed out
[  870.343951] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_set_operation_mode:1858] Failed to set driver handler
[  870.343982] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_netdev_cleanup]Unregistering netdev 53813024
[  870.457543] De-Initializing Locks
[  870.507350] Freeing wiphy
[  870.507367] Module_exit Done.
[  870.507395] wilc_sdio: probe of mmc1:0001:1 failed with error -5
```

Is there a way to provide reset pin number while doing modprobe like a modprobe option? or how else to set the reset pin to gpio 0?
Also irq pin is needed or can be left alone?

ajaykathat, 2020-06-19:
The reset pin can be configured via Devicetree or by modifying the default value of GPIO_NUM_RESET macro in [wilc_wlan.h](https://github.com/linux4wilc/driver/blob/wilc_linux_15_3_1/wilc/wilc_wlan.h#L153). 
Incase of SDIO,  the in-band interrupt is used for data transfer, and the IRQ pin is used for host wakeup if it goes to sleep mode.

mintu19, 2020-06-20:
I set 
```
#define MODALIAS		"WILC_SPI"
#define GPIO_NUM		0x5B
#define GPIO_NUM_CHIP_EN	94
#define GPIO_NUM_RESET		0
```

I have reset connected to GPIO0 as BCRM config in pi. 
 Still getting same error.
Do i need to set physical pin number here? or any other number?

mintu19, 2020-06-20:
I moved past that error by setting all gpio pins. Driver starts to load now but gets error.
I have connected en to 3.3v direct and using dummy pin in driver. I have not connected IRQ and using dummy pin for that as well.
Are these pins necessary to connect?

Logs:
```
[  671.825508] wilc_sdio: module is from the staging directory, the quality is unknown, you have been warned.
[  671.836502] Registering wifi device
[  671.836519] Max scan ids= 10,Max scan IE len= 1000,Signal Type= 1,Interface Modes= 844
[  671.837048] Initializing Locks ...
[  671.848865] wifi_pm : 0
[  671.848896] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[  671.848925] wifi_pm : 1
[  671.848935] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[  671.854328] wilc_sdio mmc1:0001:1: Driver Initializing success
[  672.009790] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.009829] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.009862] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.009893] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.009925] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.009957] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.009988] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.010020] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.010055] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.010088] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.010120] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.010151] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.010182] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.010214] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.010245] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.010278] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.010309] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.010341] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.010862] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[f89f741e] wlan0
[  672.010866] WILC POWER UP
[  672.010874] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[f89f741e][4e03a224]
[  672.010883] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[  672.010889] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[  672.010896] wilc_sdio mmc1:0001:1: SDIO speed: 50000000
[  672.010983] wilc_sdio mmc1:0001:1: wilc_sdio_cmd52..failed, err(-110)
[  672.010988] wilc_sdio mmc1:0001:1: Fail cmd 52, enable csa...
[  672.010995] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:905] Initializing WILC_Wlan FAILED
[  672.011001] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open:1029] Failed to initialize wilc
[  672.011180] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011186] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011192] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011197] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011203] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011209] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011214] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011220] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011226] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011231] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011237] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011242] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011248] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011254] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011259] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011265] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011271] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011276] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  672.011812] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[f89f741e] wlan0
[  672.011816] power up request for already powered up source Wifi
[  672.011819] Device already up. request source is Wifi
[... cut ...]
[  689.297430] wilc_sdio mmc1:0001:1: Failed cmd53, read reg (003b0000)...
[  689.297437] [wilc start]: fail read reg 0x3b0000
[  692.518905] wilc_sdio mmc1:0001:1: wilc_sdio_cmd53..failed, err(-110)
[  692.518918] wilc_sdio mmc1:0001:1: Failed cmd53, read reg (00001000)...
[  692.518928] wilc_sdio mmc1:0001:1: Unsupported chipid: 0
[  692.518944] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:905] Initializing WILC_Wlan FAILED
[  692.518956] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open:1029] Failed to initialize wilc
```

mintu19, 2020-06-22:
still no luck with driver

ajaykathat, 2020-06-22:
For WILC power-up, it's necessary to have CHIP_EN and RESET connected with the host to trigger the correct power sequence for the device.  Please refer "ATWILC Power Control" section of user guide for details about the power-up sequence. 
You can try to reload the driver after rebooting the host and device by unplugging the power.
But I recommend to connect the GPIO as suggested in the user guide to have correct power sequence for WILC device.

mintu19, 2020-06-25:
I got one step further by connecting cpi_en pin but now getting 

```
$ sudo dmesg
[   61.155634] wilc_sdio: module is from the staging directory, the quality is unknown, you have been warned.
[   61.167385] Registering wifi device
[   61.167403] Max scan ids= 10,Max scan IE len= 1000,Signal Type= 1,Interface Modes= 844
[   61.167839] Initializing Locks ...
[   61.169261] wifi_pm : 0
[   61.169289] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[   61.169319] wifi_pm : 1
[   61.169329] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[   61.175449] wilc_sdio mmc1:0001:1: Driver Initializing success
[   61.382462] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.382500] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.382533] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.382565] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.382596] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.382628] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.382660] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.382692] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.382726] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.382759] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.382791] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.382822] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.382854] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.382886] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.382918] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.382949] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.382982] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.383020] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   61.383548] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[(ptrval)] wlan0
[   61.383552] WILC POWER UP
[   61.383559] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[(ptrval)][(ptrval)]
[   61.383568] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[   61.383574] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[   61.383581] wilc_sdio mmc1:0001:1: SDIO speed: 50000000
[   66.752845] wilc_sdio mmc1:0001:1: wilc_sdio_cmd53..failed, err(-110)
[   66.752859] wilc_sdio mmc1:0001:1: Failed cmd53, read reg (003b0000)...
[   66.752866] [wilc start]: fail read reg 0x3b0000
[   72.121886] wilc_sdio mmc1:0001:1: wilc_sdio_cmd53..failed, err(-110)
[   72.121898] wilc_sdio mmc1:0001:1: Failed cmd53, read reg (00001000)...
[   72.121909] wilc_sdio mmc1:0001:1: Unsupported chipid: 0
[   72.121925] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:905] Initializing WILC_Wlan FAILED
[   72.121937] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open:1029] Failed to initialize wilc
[   72.122637] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.122651] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.122666] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.122677] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.122688] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.122700] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.122711] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.122725] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.122737] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.122748] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.122759] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.122771] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.122782] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.122793] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.122815] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.122837] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.122850] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.122861] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[   72.124032] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[(ptrval)] wlan0
[... cut ...]
[   88.310080] [wilc start]: fail read reg 0x3b0000
[   93.679087] wilc_sdio mmc1:0001:1: wilc_sdio_cmd53..failed, err(-110)
[   93.679096] wilc_sdio mmc1:0001:1: Failed cmd53, read reg (00001000)...
[   93.679102] wilc_sdio mmc1:0001:1: Unsupported chipid: 0
[   93.679111] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:905] Initializing WILC_Wlan FAILED
[   93.679117] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open:1029] Failed to initialize wilc
```

wilc_sdio_cmd53..failed, err(-110)

mintu19, 2020-06-25:
Moved one step forward by connecting irq pin as well. Now chip id is detected. Still no luck with wifi

New logs: 

```
$ sudo dmesg
[  276.495132] wilc_sdio: module is from the staging directory, the quality is unknown, you have been warned.
[  276.506181] Registering wifi device
[  276.506198] Max scan ids= 10,Max scan IE len= 1000,Signal Type= 1,Interface Modes= 844
[  276.506659] Initializing Locks ...
[  276.508231] wifi_pm : 0
[  276.508261] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[  276.508288] wifi_pm : 1
[  276.508297] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[  276.517523] wilc_sdio mmc1:0001:1: Driver Initializing success
[  276.683528] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.683565] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.683598] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.683630] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.683662] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.683694] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.683726] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.683757] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.683792] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.683826] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.683857] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.683889] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.683920] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.683952] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.683984] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.684016] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.684048] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.684080] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.684605] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[(ptrval)] wlan0
[  276.684609] WILC POWER UP
[  276.684616] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[(ptrval)][(ptrval)]
[  276.684624] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[  276.684630] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[  276.684637] wilc_sdio mmc1:0001:1: SDIO speed: 50000000
[  276.684728] wilc_sdio mmc1:0001:1: wilc_sdio_cmd52..failed, err(-110)
[  276.684732] wilc_sdio mmc1:0001:1: Fail cmd 52, enable csa...
[  276.684739] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:905] Initializing WILC_Wlan FAILED
[  276.684745] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open:1029] Failed to initialize wilc
[  276.684928] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.684934] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.684940] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.684945] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.684950] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.684956] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.684961] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.684966] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.684972] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.684977] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.684983] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.684988] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.684993] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.684999] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.685004] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.685010] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.685015] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.685020] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  276.685560] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[(ptrval)] wlan0
[  276.685564] power up request for already powered up source Wifi
[  276.685568] Device already up. request source is Wifi
[  276.685574] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[(ptrval)][(ptrval)]
[... cut ...]
[  279.504718] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinit debug Thread
[  280.557076] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[  280.557150] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:985] WLAN initialization FAILED
[  280.557162] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open:1029] Failed to initialize wilc
```

Not able to resolve firmware not available. Was it problem in building drivers? or in sending firmware to device?

mintu19, 2020-06-25:
I was able to make it work.

Steps: 
1) Download drivers and replace in kernel source.
2) enable building driver in menuconfig
3) change reset, en and irq pins as my own and set it in wilc_wlan.h file (GPIO_* as name)
4) build and install kernel
5) download firmware and paste it in /lib/firmware/mchp

closed 2020-06-25 by mintu19