linux4wilc/driver #95: WILC[13]000 hangs with Mikrotik AP

devik2, 2020-06-24:
Hello,
we used both WILC1000 and WILC3000 in our linux based (STM32MP1 MPU) designs.
The module is connected via SPI at 40Mbit in both designs.
Both designs have similar problem connectin to our Mikrotik AP (and connects to some
other APs ok). After connection attempt the WILC is left in highly unstable state until its 
hw reset. 
There is separate problem of WILC_DEBUG kernel thread taking 90% of CPU all the time,
but this one is minor issue just now.
We use latest driver code from git and 15.4 firmware. I attached whole dmesg and also
wpa_supplicant output. Conn attemtps starts at times 7309 and 7426.
Any idea ?
Thanks, Martin

[wilc.txt](https://github.com/linux4wilc/driver/files/4825681/wilc.txt)
[dmsg.txt](https://github.com/linux4wilc/driver/files/4825699/dmsg.txt)

ajaykathat, 2020-06-25:
The WILC_DEBUG logs can be enabled or disabled at run time by following instructions from "Debug log" section of [user guide.](http://ww1.microchip.com/downloads/en/DeviceDoc/ATWILC1000-ATWILC3000-Wi-Fi-Link-Controller-Linux-User-Guide-DS70005328B.pdf). Also, the default value for the debug level can be changed from code by modifying WILC_DEBUG_REGION macro. 

For the connect issue, the device tries to connect with 2 different SSID(devix2/Android) one after the other. Please check "mod/wpa.conf" first by configuring to connect with a single network.  It seems that connection is terminated by wpa-supplicant with Android SSID also. So I suggest first try to configure to connect with single SSID and check if .conf files parameters are fine. If possible please capture the WIFI sniffer logs which might help to understand the exact reason for connection timeout.

Alternatively "wpa_cli" commands can be tried to configure a connection with specific AP for that please refer to the "Running in the WILC station mode" section it helps to configure and connect with AP.

devik2, 2020-06-26:
Thank you for the reply. Good to know I can disable the logs.
Regarding connect issue, I tried both. In majority of tests I have only single network
section in conf (devix2 one).
The termination (seen in Android too) was caused by terminating wpa_supplicant by SIGINT.
In case of devix2 network it dies early by: wlan0: Authentication with 6c:3b:6b:52:35:17 timed out.

Wifi-sniffer logs .. How to do it ? Are you speaking about extra wireless sniffer device or can I somehow
tap WILC device ?

ajaykathat, 2020-07-03:
For capturing wireless logs you have to use an external WiFi sniffer device. 
Maybe you can try connecting with devix2  with different security to check if it works. 
Btw I am not clear how wpa_ssupplicant is picking different SSID for next time. Are you closing 'wps_s' first and then changing the SSID to start with devix2 SSID. What exactly is the complete test procedure and setup?

closed 2021-03-21 by devik2