linux4wilc/driver #96: driver wilc1000 for kernel 4.14.98

DomenicoAcri, 2020-07-02:
Hi,
I'm using an NXP imx8mmini processor, and we connected the ATWILC1000 module on the SDIO interface. 
Where can I find drivers and firmware for kernel version 4.14.98?

Best Regards
Domenico Acri

ajaykathat, 2020-07-03:
FW can be downloaded from https://github.com/linux4wilc/firmware repo.  For wilc1000, use wilc1000_wifi_firmware.bin FW binary.

The link to 15.3.1 driver & FW is below:
FW-> https://github.com/linux4wilc/firmware/tree/wilc_linux_15_3_1
Driver -> https://github.com/linux4wilc/driver/tree/wilc_linux_15_3_1

DomenicoAcri, 2020-07-03:
Hi,
thanks a lot for the answer.
I integrated the driver and firmware, but it gives this error:

insmod wilc-sdio.ko 
[   71.928370] Registering wifi device
[   71.931969] Max scan ids= 10,Max scan IE len= 1000,Signal Type= 1,Interface Modes= 844
[   71.940135] Initializing Locks ...
[   71.945659] wifi_pm : 0
[   71.948149] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[   71.955071] wifi_pm : 1
[   71.957595] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[   71.971026] wilc_sdio mmc1:0001:1: Driver Initializing success

ifconfig wlan0 up
[   87.193039] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[ffff8000109fc000] wlan0
[   87.201638] WILC POWER UP
[   87.204264] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[ffff8000109fc000][ffff8000109fdce8]
[   87.213954] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[   87.221038] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[   87.228816] wilc_sdio mmc1:0001:1: SDIO speed: 25000000
[   87.242136] wilc_sdio mmc1:0001:1: Unsupported chipid: 0
[   87.247512] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:905] Initializing WILC_Wlan FAILED
[   87.256764] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open:1029] Failed to initialize wilc
SIOCSIFFLAGS: Input/output error

This is my device tree:

/* WIFI */
&usdhc2 {
	pinctrl-names = "default";
	pinctrl-0 = <&pinctrl_usdhc2>, <&pinctrl_atwilc1000>;
	bus-width = <4>;
	no-1-8-v;
	max-frequency = <25000000>;
	non-removable;
	status = "okay";

	wilc_sdio@0{
			compatible = "microchip,wilc1000", "microchip,wilc3000";
			reset-gpios = <&gpio1 10 GPIO_ACTIVE_HIGH>;
			chip_en-gpios = <&gpio1 3 GPIO_ACTIVE_HIGH>;
			irq-gpios = <&gpio1 8 GPIO_ACTIVE_HIGH>;
			status = "okay";
			reg = <0>;
			bus-width = <4>;
	};
};

pinctrl_atwilc1000: atwilc1000grp {
    fsl,pins = <
        MX8MM_IOMUXC_GPIO1_IO03_GPIO1_IO3 0x19 /* WIFI_REGON */
        MX8MM_IOMUXC_GPIO1_IO10_GPIO1_IO10 0x19 /* WIFI_NRST */
        MX8MM_IOMUXC_GPIO1_IO08_GPIO1_IO8 0x1C0 /* WIFI_IRQN */
    >;
};

Best Regards
Domenico Acri

ajaykathat, 2020-07-03:
From the logs, it appears that the driver fails to get 'reset' GPIO using gpiod_get() API.  During failure, it tries to load a default GPIO 'GPIO_NUM_RESET' which may be not correct in your codebase. 

"wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default"

Please check and try by configuring with correct available GPIO for reset.

DomenicoAcri, 2020-07-03:
Hi,
yes I suspected it. 
So I went to change the default GPIOs in the wilc_wlan.h file as follows:

#define GPIO_NUM 8
#define GPIO_NUM_CHIP_EN 3
#define GPIO_NUM_RESET 10

this is the output:

insmod wilc-sdio.ko 
[  100.647908] Registering wifi device
[  100.651410] Max scan ids= 10,Max scan IE len= 1000,Signal Type= 1,Interface Modes= 844
[  100.659584] Initializing Locks ...
[  100.665204] wifi_pm : 0
[  100.667688] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[  100.674623] wifi_pm : 1
[  100.677162] wilc_sdio mmc1:0001:1: succesfully got gpio_reset
[  100.683088] wilc_sdio mmc1:0001:1: succesfully got gpio_chip_en
[  100.695940] wilc_sdio mmc1:0001:1: Driver Initializing success
root@imx8mm-alma:~# ifconfig wlan0 up
[  108.771614] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[ffff800010974000] wlan0
[  108.780190] WILC POWER UP
[  108.782815] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[ffff800010974000][ffff800010975ce8]
[  108.792510] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[  108.799597] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[  108.807379] wilc_sdio mmc1:0001:1: SDIO speed: 25000000
[  108.820286] wilc_sdio mmc1:0001:1: wilc_sdio_cmd52..failed, err(-110)
[  108.826747] wilc_sdio mmc1:0001:1: Fail cmd 52, enable csa...
[  108.832563] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:905] Initializing WILC_Wlan FAILED
[  108.841823] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open:1029] Failed to initialize wilc
SIOCSIFFLAGS: Input/output error

there still seems to be a problem.

Best Regards
Domenico Acri

ajaykathat, 2020-07-03:
The logs look a bit strange for fetching reset GPIO because the next call for wilc_wlan_power() it was able to fetch reset from your Device tree. The below logs were missing for the next call to wilc_wlan_power().

[ 100.665204] wifi_pm : 0
[ 100.667688] wilc_sdio mmc1:0001:1: failed to get Reset GPIO, try default
[ 100.674623] wifi_pm : 1

So still I suspect the reset GPIO is configured properly for the proper pwr sequence of WILC device.

Can you try commenting below code in wilc_wlan_power_on_sequence() to check the behaviour?
```
int wilc_wlan_power_on_sequence(struct wilc *wilc)
{
#if 0
	int ret;

	ret = wilc_wlan_power(wilc, 0);
	if (ret)
		return ret;
	ret = wilc_wlan_power(wilc, 1);
	if (ret)
		return ret;
#endif
	return 0;
}
```

DomenicoAcri, 2020-07-06:
Hi,
with the last modification this is the output:

~# insmod wilc-sdio.ko 
[   18.901985] Registering wifi device
[   18.905506] Max scan ids= 10,Max scan IE len= 1000,Signal Type= 1,Interface Modes= 844
[   18.913672] Initializing Locks ...
[   18.918858] wilc_sdio mmc1:0001:1: Driver Initializing success

~# ifconfig wlan0 up
[   25.608816] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[ffff8000102cc000] wlan0
[   25.617397] WILC POWER UP
[   25.620059] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[ffff8000102cc000][ffff8000102cdce8]
[   25.629745] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[   25.636826] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[   25.644651] wilc_sdio mmc1:0001:1: SDIO speed: 25000000
[   25.657920] wilc_sdio mmc1:0001:1: Unsupported chipid: 0
[   25.663262] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:905] Initializing WILC_Wlan FAILED
[   25.672512] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open:1029] Failed to initialize wilc
SIOCSIFFLAGS: Input/output error

Best Regards
Domenico Acri

ajaykathat, 2020-07-07:
I suspect the issue could be related to power_sequence not triggered from host to WILC device. I hope you have tried by rebooting the host by removing power cable too. 
If you still observe the issue, I suggest submitting a case in [salesforce](https://microchipsupport.force.com/s/article/How-to-submit-a-case) with host setup details used to connect WILC device.