linux4wilc/driver #98: Kernel Panik on insmod

gumulka, 2020-07-09:
I've compiled the driver for Kernel 4.14.156 under OpenWRT and when trying to insert the SPI module, I get a Kernel Panik.

Source version is tag `wilc_linux_15_3_1`

```
[  279.571320] ------------[ cut here ]------------
[  279.576100] WARNING: CPU: 0 PID: 3716 at backports-4.19.85-1/net/wireless/core.c:743 wiphy_register+0x2f0/0xa7c [cfg80211]
[  279.587274] Modules linked in: wilc_spi(+) mac80211 cfg80211 compat pppoe ppp_async option l2tp_ppp cdc_mbim usb_wwan qmi_wwan pppox ppps
[  279.658286]  flexcan crc_ccitt cdc_wdm can_raw can_dev can fuse nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6d
[  279.729422]  mc100_gpio mii
[  279.732278] CPU: 0 PID: 3716 Comm: modprobe Not tainted 4.14.156 #0
[  279.738559] Hardware name: Freescale i.MX6 Ultralite (Device Tree)
[  279.744748] Backtrace: 
[  279.747235] [<8010b158>] (dump_backtrace) from [<8010b44c>] (show_stack+0x18/0x1c)
[  279.754827]  r7:7f683888 r6:60000013 r5:00000000 r4:80a3a760
[  279.760517] [<8010b434>] (show_stack) from [<80725f3c>] (dump_stack+0x9c/0xb0)
[  279.767769] [<80725ea0>] (dump_stack) from [<80129f74>] (__warn+0xec/0x108)
[  279.774751]  r7:7f683888 r6:7f6b57a4 r5:00000000 r4:00000000
[  279.780433] [<80129e88>] (__warn) from [<8012a048>] (warn_slowpath_null+0x28/0x30)
[  279.788023]  r9:0000034c r8:00000003 r7:00000000 r6:00000000 r5:9e18e9c0 r4:9e18e1a0
[  279.795887] [<8012a020>] (warn_slowpath_null) from [<7f683888>] (wiphy_register+0x2f0/0xa7c [cfg80211])
[  279.805497] [<7f683598>] (wiphy_register [cfg80211]) from [<7f754364>] (wilc_create_wiphy+0x17c/0x1b4 [wilc_spi])
[  279.815783]  r10:000001f4 r9:0000034c r8:00000003 r7:9f773800 r6:00000005 r5:9e18e440
[  279.823623]  r4:9e18e1a0
[  279.826352] [<7f7541e8>] (wilc_create_wiphy [wilc_spi]) from [<7f7543c0>] (wilc_cfg80211_init+0x24/0x2c8 [wilc_spi])
[  279.836900]  r10:00000000 r9:9e3bfc7c r8:00000001 r7:7f76430c r6:00000000 r5:7f76a1f8
[  279.844739]  r4:9f773800
[  279.847461] [<7f75439c>] (wilc_cfg80211_init [wilc_spi]) from [<7f76103c>] (wilc_bus_probe+0x60/0xa50 [wilc_spi])
[  279.857747]  r9:00000028 r8:7f76a208 r7:9e949240 r6:00000000 r5:7f76a1f8 r4:9f773800
[  279.865602] [<7f760fdc>] (wilc_bus_probe [wilc_spi]) from [<804fd9e0>] (spi_drv_probe+0x8c/0xa8)
[  279.874406]  r7:00000000 r6:00000000 r5:7f76a1f8 r4:9f773800
[  279.880090] [<804fd954>] (spi_drv_probe) from [<80477534>] (driver_probe_device+0x150/0x2e8)
[  279.888544]  r7:00000000 r6:80a6e378 r5:80a6e374 r4:9f773800
[  279.894223] [<804773e4>] (driver_probe_device) from [<80477754>] (__driver_attach+0x88/0xac)
[  279.902682]  r10:80a03c08 r9:014000c0 r8:7f770000 r7:00000000 r6:9f773834 r5:7f76a208
[  279.910523]  r4:9f773800 r3:00000000
[  279.914124] [<804776cc>] (__driver_attach) from [<80475974>] (bus_for_each_dev+0x54/0xa8)
[  279.922319]  r7:00000000 r6:804776cc r5:7f76a208 r4:00000000
[  279.928005] [<80475920>] (bus_for_each_dev) from [<80476fa0>] (driver_attach+0x24/0x28)
[  279.936024]  r6:80a229b0 r5:9e30d400 r4:7f76a208
[  279.940666] [<80476f7c>] (driver_attach) from [<80476a50>] (bus_add_driver+0xf0/0x208)
[  279.948607] [<80476960>] (bus_add_driver) from [<80477ea4>] (driver_register+0xb0/0xec)
[  279.956628]  r7:9e949c80 r6:ffffe000 r5:00000000 r4:7f76a208
[  279.962309] [<80477df4>] (driver_register) from [<804fd914>] (__spi_register_driver+0x60/0x64)
[  279.970932]  r5:00000000 r4:80a03c08
[  279.974625] [<804fd8b4>] (__spi_register_driver) from [<7f770020>] (init_module+0x20/0x1000 [wilc_spi])
[  279.984130] [<7f770000>] (init_module [wilc_spi]) from [<80101a90>] (do_one_initcall+0xc8/0x18c)
[  279.992947] [<801019c8>] (do_one_initcall) from [<801969c8>] (do_init_module+0x68/0x1f4)
[  280.001062]  r9:014000c0 r8:7f76a280 r7:9e949c80 r6:9e374980 r5:9e374800 r4:7f76a280
[  280.008830] [<80196960>] (do_init_module) from [<80195d0c>] (load_module+0x1e38/0x21d8)
[  280.016851]  r7:00000000 r6:9e374980 r5:9e374800 r4:9e3bff30
[  280.022536] [<80193ed4>] (load_module) from [<801961f0>] (SyS_init_module+0x144/0x17c)
[  280.030474]  r10:00000051 r9:000128de r8:ffffe000 r7:a0eaa8f4 r6:76f09904 r5:00000000
[  280.038312]  r4:000078f4
[  280.040872] [<801960ac>] (SyS_init_module) from [<80107a60>] (ret_fast_syscall+0x0/0x54)
[  280.048985]  r10:00000080 r9:9e3be000 r8:80107c64 r7:00000080 r6:00000003 r5:00000000
[  280.056824]  r4:00000000
[  280.059464] ---[ end trace b5b9cd23ea4bfed2 ]---
[  280.064136] ------------[ cut here ]------------
[  280.068881] WARNING: CPU: 0) from [<7f754364>] (wilc_create_wiphy+0x17c/0x1b4 [wilc_spi])
[  280.309661]  r10:000001f4 r9:0000034c r8:00000003 r7:9f773800 r6:00000005 r5:9e18e440
[  280.317503]  r4:9e18e1a0
[  280.320234] [<7f7541e8>] (wilc_create_wiphy [wilc_spi]) from [<7f7543c0>] (wilc_cfg80211_init+0x24/0x2c8 [wilc_spi])
[  280.330781]  r10:00000000 r9:9e3bfc7c r8:00000001 r7:7f76430c r6:00000000 r5:7f76a1f8
[  280.338621]  r4:9f773800
[  280.341341] [<7f75439c>] (wilc_cfg80211_init [wilc_spi]) from [<7f76103c>] (wilc_bus_probe+0x60/0xa50 [wilc_spi])
[  280.351625]  r9:00000028 r8:7f76a208 r7:9e949240 r6:00000000 r5:7f76a1f8 r4:9f773800
[  280.359482] [<7f760fdc>] (wilc_bus_probe [wilc_spi]) from [<804fd9e0>] (spi_drv_probe+0x8c/0xa8)
[  280.368285]  r7:00000000 r6:00000000 r5:7f76a1f8 r4:9f773800
[  280.373968] [<804fd954>] (spi_drv_probe) from [<80477534>] (driver_probe_device+0x150/0x2e8)
[... 64 lines cut ...]
[  281.151310]  r7:00000000 r6:804776cc r5:7f76a208 r4:00000000
[  281.156995] [<80475920>] (bus_for_each_dev) from [<80476fa0>] (driver_attach+0x24/0x28)
[  281.165015]  r6:80a229b0 r5:9e30d400 r4:7f76a208
[  281.169658] [<80476f7c>] (driver_attach) from [<80476a50>] (bus_add_driver+0xf0/0x208)
[  281.177598] [<80476960>] (bus_add_driver) from [<80477ea4>] (driver_register+0xb0/0xec)
[  281.185618]  r7:9e949c80 r6:ffffe000 r5:00000000 r4:7f76a208
[  281.191299] [<80477df4>] (driver_register) from [<804fd914>] (__spi_register_driver+0x60/0x64)
[  281.199924]  r5:00000000 r4:80a03c08
[  281.203611] [<804fd8b4>] (__spi_register_driver) from [<7f770020>] (init_module+0x20/0x1000 [wilc_spi])
[  281.213113] [<7f770000>] (init_module [wilc_spi]) from [<80101a90>] (do_one_initcall+0xc8/0x18c)
[  281.221926] [<801019c8>] (do_one_initcall) from [<801969c8>] (do_init_module+0x68/0x1f4)
[  281.230040]  r9:014000c0 r8:7f76a280 r7:9e949c80 r6:9e374980 r5:9e374800 r4:7f76a280
[  281.237806] [<80196960>] (do_init_module) from [<80195d0c>] (load_module+0x1e38/0x21d8)
[  281.245826]  r7:00000000 r6:9e374980 r5:9e374800 r4:9e3bff30
[  281.251512] [<80193ed4>] (load_module) from [<801961f0>] (SyS_init_module+0x144/0x17c)
[  281.259449]  r10:00000051 r9:000128de r8:ffffe000 r7:a0eaa8f4 r6:76f09904 r5:00000000
[  281.267289]  r4:000078f4
[  281.269848] [<801960ac>] (SyS_init_module) from [<80107a60>] (ret_fast_syscall+0x0/0x54)
[  281.277959]  r10:00000080 r9:9e3be000 r8:80107c64 r7:00000080 r6:00000003 r5:00000000
[  281.285800]  r4:00000000
[  281.288354] Code: e92dd800 e24cb004 e3a03000 e280200c (e5803000) 
[  281.294556] ---[ end trace b5b9cd23ea4bfed4 ]---
[  281.299195] Kernel panic - not syncing: Fatal exception
[  281.304444] Rebooting in 3 seconds..
```

ajaykathat, 2020-07-10:
From the crash dump, it looks like wiphy_register() is failed because of n_iface_combinations parameter. In WILC driver, doesn't configure 'n_iface_combinations' parameter during the initialization. 

Could you please check in the driver code is modified to set the n_iface_combinations in the code. 

The above understanding is based on the below functions mapping from the stack dump.

backports-4.19.85-1/net/wireless/core.c + 743 
https://elixir.bootlin.com/linux/v4.19.85/source/net/wireless/core.c#L743
  |
  v
https://elixir.bootlin.com/linux/v4.19.85/source/net/wireless/core.c#L536

wiphy_verify_combinations() return failure only when incorrect n_iface_combinations is defined. n_iface_combinations should have value zero(as wilc driver doesn't set this value).

gumulka, 2020-07-10:
Not exactly sure, what you are looking for, but here is the core.c file: https://gist.github.com/gumulka/528d5c67fd560ed3f3be5ae5ce749072

From what I can see neither does core.c set n_iface_combinations, but does use it. 

Do I have to set it somehow somewhere and if yes, where?

ajaykathat, 2020-07-10:
The dump shows warning at +743 line "backports-4.19.85-1/net/wireless/core.c:743" . In your shared core.c file it contains different code at line#743. Check for the possible reason why wiphy_register() fails in your setup.

gumulka, 2020-07-10:
The code I shared, had a WARN_ON on line [743](https://gist.github.com/gumulka/528d5c67fd560ed3f3be5ae5ce749072#file-core-c-L743). Isn't WARN_ON supposed to produce this kind of first error message?

The WARN in case is produced, because wiphy->interface_modes is never set (see https://github.com/linux4wilc/driver/blob/master/wilc/wilc_wfi_cfgoperations.c#L2483 ), which in case seems to be because none of the modes are defined. Or at least I could not find their definitions anywhere. 

My command to compile the wilc driver into a ko:

```
make -C "/home/pflug/temp/OpenWRT/build_dir/target-arm_cortex-a9+neon_musl_eabi/linux-imx6/linux-4.14.156"  \
  HOSTCFLAGS="-O2 -I/home/pflug/temp/OpenWRT/staging_dir/host/include -I/home/pflug/temp/OpenWRT/staging_dir/hostpkg/include -I/home/pflug/temp/OpenWRT/staging_dir/target-arm_cortex-a9+neon_musl_eabi/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" \
  CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" \
  ARCH="arm" \
  KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Nov 30 15:52:33 2019" KBUILD_BUILD_VERSION="0" \
  HOST_LOADLIBES="-L/home/pflug/temp/OpenWRT/staging_dir/host/lib" \
  CONFIG_SHELL="bash" V=''  cmd_syscalls= \
  KERNELRELEASE=4.14.156 \
  SUBDIRS="/home/pflug/temp/OpenWRT/build_dir/target-arm_cortex-a9+neon_musl_eabi/linux-imx6/wilc1000-15.3.1/wilc" \
  CONFIG_WILC_SPI="m" \
  modules
```

whereas `/home/pflug/temp/OpenWRT/build_dir/target-arm_cortex-a9+neon_musl_eabi/linux-imx6/wilc1000-15.3.1/` is this repository, checked out to tag `wilc_linux_15_3_1`

Is there another way to compile the wilc driver into a loadable module, that I'm not aware of?

gumulka, 2020-07-10:
Okay, I think I've got an idea. Could it be because the max80211 is compiled with:

```
make  -C "/home/pflug/temp/OpenWRT/build_dir/target-arm_cortex-a9+neon_musl_eabi/linux-imx6/backports-4.19.85-1" \
  HOSTCFLAGS="-O2 -I/home/pflug/temp/OpenWRT/staging_dir/host/include -I/home/pflug/temp/OpenWRT/staging_dir/hostpkg/include -I/home/pflug/temp/OpenWRT/staging_dir/target-arm_cortex-a9+neon_musl_eabi/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" \
  CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" \
  ARCH="arm" \
  KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sat Nov 30 15:52:33 2019" KBUILD_BUILD_VERSION="0" \
  HOST_LOADLIBES="-L/home/pflug/temp/OpenWRT/staging_dir/host/lib" \
  CONFIG_SHELL="bash" V=''  cmd_syscalls= \
  KERNELRELEASE=4.14.156 \
  EXTRA_CFLAGS="-I/home/pflug/temp/OpenWRT/build_dir/target-arm_cortex-a9+neon_musl_eabi/linux-imx6/backports-4.19.85-1/include -iremap/home/pflug/temp/OpenWRT/build_dir/target-arm_cortex-a9+neon_musl_eabi/linux-imx6/backports-4.19.85-1:backports-4.19.85-1" \
  KLIB_BUILD="/home/pflug/temp/OpenWRT/build_dir/target-arm_cortex-a9+neon_musl_eabi/linux-imx6/linux-4.14.156" \
  MODPROBE=true KLIB=/lib/modules/4.14.156 KERNEL_SUBLEVEL=14 \
  KBUILD_LDFLAGS_MODULE_PREREQ= \
  modules
```
That the wilc and mac are not compatible?

But how do I change my compiling to insert the wilc correctly?

ajaykathat, 2020-07-10:
Yes, As I see, it looks like there is a mismatch between the cfg80211 subsystem and wilc1000 driver code is compiled against.  The WILC driver is setting the supported interface modes(interface_modes) in the following line(https://github.com/linux4wilc/driver/blob/master/wilc/wilc_wfi_cfgoperations.c#L2483 ) but still core.c file reports incorrect warning of value not set.

If you notice the stack dump it looks like kernel version is 4.19.85-1
**"backports-4.19.85-1/net/wireless/core.c:743"**

 but the driver module is compiled against linux-4.14.156. 
"Comm: modprobe Not tainted 4.14.156"

gumulka, 2020-07-13:
If anybody else is having this issue with OpenWRT, I got it working, by patching and extending the mac80211 package with the wilc code and letting it compile the driver.

closed 2020-07-13 by gumulka

kishangondaliya, 2021-06-24:
@gumulka can you please share your patch if possible. I'm also facing same issue.

gumulka, 2021-06-24:
@kishangondaliya I don't have the code with me at the moment, but in general:

* copy the backports directory from `build_dir/target..../linux-XXX/backports-XXX` into a new directory outside of OpenWRT
* go there, call git init and add everything there is
* then copy the code from this repository to the appropriate folder and change the Makefile to also compile the wilc folder
* in OpenWRT `package/kernel/mac80211/patches/`  create directory wilc and 
* create a patch from your backports directory and put this patch inside the newly created wilc directory
* next append the wilc patch dir to the `Build/Patch` target in `package/kernel/mac80211/Makefile`
* create in this makefile a target for wilc. You can copy and modify one of the existing ones.
* Lastly add the package to your .config and compile OpenWRT

I hope this helps

kishangondaliya, 2021-06-25:
@gumulka thanks for the instructions. I'll try this out.