linux4wilc/driver #99: Module does not send /recv

gumulka, 2020-07-16:
Whatever I do, I can't get the module to register any WLAN in the proximity, although there are several.

My dmesg from start to scan:
```
[   61.372323] Loading modules backported from Linux version v4.19.85-0-gc63ee2939dc1
[   61.372337] Backport generated by backports.git v4.19.85-1-0-g8a8be258
[   61.413710] WILC_SPI spi2.0: spiModalias: wilc1000, spiMax-Speed: 48000000
[   61.413752] Registering wifi device
[   61.413769] Max scan ids= 10,Max scan IE len= 1000,Signal Type= 1,Interface Modes= 844
[   61.414267] Initializing Locks ...
[   61.426612] wifi_pm : 0
[   61.426699] WILC_SPI spi2.0: succesfully got gpio_reset
[   61.426734] WILC_SPI spi2.0: succesfully got gpio_chip_en
[   61.426764] wifi_pm : 1
[   61.426797] WILC_SPI spi2.0: succesfully got gpio_reset
[   61.426829] WILC_SPI spi2.0: succesfully got gpio_chip_en
[   61.432417] WILC_SPI spi2.0: WILC SPI probe success
[   76.801252] WILC_SPI spi2.0 wlan0: INFO [wilc_mac_open]MAC OPEN[9f4d6000] wlan0
[   76.801270] WILC POWER UP
[   76.801289] WILC_SPI spi2.0 wlan0: INFO [wilc_init_host_int]Host[9f4d6000][9f4d7520]
[   76.801313] WILC_SPI spi2.0 wlan0: INFO [wilc_mac_open]*** re-init ***
[   76.801328] WILC_SPI spi2.0 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[   76.803564] WILC_SPI spi2.0 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
[   76.803585] WILC_SPI spi2.0 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
[   76.803598] WILC_SPI spi2.0 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
[   76.803811] WILC_SPI spi2.0 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
[   76.804032] WILC_SPI spi2.0: got gpio_irq successfully
[   76.804348] WILC_SPI spi2.0 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 172
[   76.804374] WILC_SPI spi2.0 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC1000
[   76.804389] WILC_SPI spi2.0 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc1000_wifi_firmware.bin
[   76.804403] WILC_SPI spi2.0 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc1000_wifi_firmware.bin
[   76.810456] WILC_SPI spi2.0 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
[   76.810482] wilc_wlan_firmware_downloadDownloading firmware size = 135832
[   76.877927] wilc_wlan_firmware_download Offset = 120380
[   76.887060] wilc_wlan_firmware_download Offset = 135832
[   76.887090] WILC_SPI spi2.0 wlan0: INFO [wilc_firmware_download]Download Succeeded
[   76.887145] WILC_SPI spi2.0 wlan0: INFO [wilc_start_firmware]Starting Firmware ...
[   76.888431] WILC_SPI spi2.0 wlan0: INFO [wilc_start_firmware]Waiting for FW to get ready ...
[   76.962414] cfg_indicate_rx: Info message received
[   76.962761] WILC_SPI spi2.0 wlan0: INFO [wilc_start_firmware]Firmware successfully started
[   76.972377] WILC_SPI spi2.0 wlan0: INFO [wilc_wlan_initialize]WILC Firmware Ver = WILC_WIFI_FW_REL_15_3_1 Build: 11736
[   76.972395] WILC_SPI spi2.0 wlan0: INFO [wilc_init_fw_config]Start configuring Firmware
[   77.093275] WILC_SPI spi2.0 wlan0: INFO [wilc_mac_open]Mac address: 00:50:c2:5e:10:8e
[   77.093317] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 2.
[   77.093338] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 33:33:00:00:00:01
[   77.093354] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 33:33:00:00:00:02
[   77.093759] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 3.
[   77.093783] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 33:33:00:00:00:01
[   77.093800] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 33:33:00:00:00:02
[   77.093814] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Entry[2]: 01:00:5e:00:00:01
[   77.094178] WILC_SPI spi2.0 wlan0: INFO [set_power_mgmt]dev [wlan0]
[   77.094200] WILC_SPI spi2.0 wlan0: INFO [set_power_mgmt] Power save Enabled= 0 , TimeOut = -1
[   77.112366] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 4.
[   77.112397] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 33:33:00:00:00:01
[   77.112411] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 33:33:00:00:00:02
[   77.112425] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Entry[2]: 01:00:5e:00:00:01
[   77.112440] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Entry[3]: 33:33:ff:5e:10:8e
[   78.242272] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 5.
[   78.242301] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 33:33:00:00:00:01
[   78.242316] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 33:33:00:00:00:02
[   78.242330] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Entry[2]: 01:00:5e:00:00:01
[   78.242345] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Entry[3]: 33:33:ff:5e:10:8e
[   78.242357] WILC_SPI spi2.0 wlan0: INFO [wilc_set_multicast_list]Entry[4]: 33:33:ff:00:00:00
[  370.402981] WILC_SPI spi2.0 wlan0: INFO [scan]Requested num of channel 14
[  370.403004] WILC_SPI spi2.0 wlan0: INFO [scan]Scan Request IE len =  0
[  370.403020] WILC_SPI spi2.0 wlan0: INFO [scan]Number of SSIDs 1
[  370.403033] WILC_SPI spi2.0 wlan0: INFO [scan]Trigger Scan Request
[  371.012881] cfg_indicate_rx: Scan Notification Received
[  371.012914] WILC_SPI spi2.0 wlan0: INFO [wilc_scan_complete_received]Scan notification received
[  371.013270] WILC_SPI spi2.0 wlan0: INFO [cfg_scan_result]Scan Done[9f4d6000]
```

The scan result is always empty and I'm a bit confused about the MAC-Address. Shouldn't it start with f8: ?

ajaykathat, 2020-07-22:
Please try "wpa_cli -iwlan0 scan_results" command to check the available networks after wpa_supplicant start.

The WILC driver doesn't have logs to print found network details enabled by default.  To print the actual found network from the driver please enable below logs in cfg_scan_result() by using PRINT_INFO()

```
PRINT_INFO(priv->dev, CFG80211_DBG,
			"Network Info:: CHANNEL Frequency: %d, RSSI: %d,\n",
			freq, ((s32)info->rssi * 100));

```
  
Regarding MAC address, "00:50:c2:5e:10:8e" is the default address which is assigned when the address is not loaded from OTP memory. The reason for the MAC address is not captured in driver logs. So please raise a [sale force ticket](https://microchipsupport.force.com/s/article/How-to-submit-a-case) with more information about the module.

IMO, the connect issue is not related to the MAC address. The connection failure could be because the suitable match for AP configured in wpa_supplicant is not found during the scan. 
Please check the contents of the configuration file which is specified using -c option with wpa_supplicant.

You can try steps mentioned in "8.5 Running in the ATWILC Station mode" in [UserGuide.](http://ww1.microchip.com/downloads/en/DeviceDoc/ATWILC1000,ATWILC3000-User-Guide-DS70005328D.pdf)

gumulka, 2020-08-27:
Found the issue to be hardware related. The module could not get a signal. After Fixing the Hardware failure, it worked.

closed 2020-08-27 by gumulka