linux4wilc/driver #100: ATWILC 1000 SDIO Failed to start firmware

Andrflor, 2020-07-23:
Hello,
I'm using 15.3.1 ATWILC1000 sdio on a cm3 with a V4.19.80 kernel

I have a pretty random comportment, most of the time wifi does not work, but some times it does work for some reason
It seems that it restart multiple times
I have hardcoded gpio pins in the driver header and configured sdio overlay.

I don't understand what is happening, if you have some hint :smile_cat: 

```[    4.733829] wilc_sdio: no symbol version for module_layout
[    4.777717] wilc_sdio mmc1:0001:1: succesfully got gpio_reset
[    4.777738] wilc_sdio mmc1:0001:1: succesfully got gpio_chip_en
[    4.777771] wilc_sdio mmc1:0001:1: succesfully got gpio_reset
[    4.777790] wilc_sdio mmc1:0001:1: succesfully got gpio_chip_en
[    4.794740] wilc_sdio mmc1:0001:1: Driver Initializing success
[    9.944819] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[11e0aa06] wlan0
[    9.944833] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[11e0aa06][8d002279]
[    9.944853] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[    9.944862] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[    9.944873] wilc_sdio mmc1:0001:1: SDIO speed: 25000000
[    9.945582] wilc_sdio mmc1:0001:1: chipid 001003a0
[    9.946359] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
[    9.946373] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
[    9.946380] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
[    9.946568] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
[    9.946713] wilc_sdio mmc1:0001:1: failed to get IRQ GPIO, load default
[    9.946777] wilc_sdio mmc1:0001:1 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 176
[    9.946857] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC1000
[    9.946867] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc1000_wifi_firmware.bin
[    9.946880] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc1000_wifi_firmware.bin
[    9.947192] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000000
[    9.956349] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
[    9.956362] wilc_wlan_firmware_downloadDownloading firmware size = 135832
[    9.972352] wilc_wlan_firmware_download Offset = 120380
[    9.974466] wilc_wlan_firmware_download Offset = 135832
[    9.974479] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Download Succeeded
[    9.974516] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_start_firmware]Starting Firmware ...
[    9.975170] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_start_firmware]Waiting for FW to get ready ...
[   10.088056] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_start_firmware]Firmware successfully started
[   10.097503] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Firmware Ver = WILC_WIFI_FW_REL_15_3_1 Build: 11736
[   10.097532] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_fw_config]Start configuring Firmware
[   10.217862] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]Mac address: f8:f0:05:ae:bb:13
[   10.217917] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 0.
[   10.217929] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Enable mcast filter retrive directed pkts only
[   10.218163] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 1.
[   10.218177] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 01:00:5e:00:00:01
[   10.218217] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 2.
[   10.218230] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 01:00:5e:00:00:01
[   10.218242] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 33:33:00:00:00:01
[   10.218375] wilc_sdio mmc1:0001:1 wlan0: INFO [set_power_mgmt]dev [wlan0]
[   10.218388] wilc_sdio mmc1:0001:1 wlan0: INFO [set_power_mgmt] Power save Enabled= 1 , TimeOut = -1
[   10.245698] wilc_sdio mmc1:0001:1 wlan0: INFO [get_tx_power]Got tx power 18
[   10.246493] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Setting mcast List with count = 2.
[   10.246511] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[0]: 01:00:5e:00:00:01
[   10.246523] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_set_multicast_list]Entry[1]: 33:33:00:00:00:01
[   10.246729] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_close]Mac close
[   10.302643] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_close]Deinitializing wilc
[   10.302677] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing wilc  ...
[   10.302699] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Disabling IRQ
[   10.302819] wilc_sdio mmc1:0001:1: wilc_sdio_disable_interrupt
[   10.302913] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing Threads
[   10.302925] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[   10.302936] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinit debug Thread
[   11.031824] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[   11.031854] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing IRQ
[   11.036809] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]Deinitializing WILC Wlan
[   11.036830] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_deinitialize]wilc deinitialization Done
[   11.037924] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_send_config_pkt:1799] Get Timed out
[   11.037941] wilc_sdio mmc1:0001:1 wlan0: Failed to get mac address
[   11.037950] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_send_config_pkt:1816] Set Timed out
[... 324 lines cut ...]
[  993.089924] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_initialize]WILC Initialization done
[  993.089944] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Initializing Threads ...
[  993.089954] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for transmission
[  993.090219] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_initialize_threads]Creating kthread for Debugging
[  993.090410] wilc_sdio mmc1:0001:1: got gpio_irq successfully
[  993.090463] wilc_sdio mmc1:0001:1 wlan0: INFO [init_irq]IRQ request succeeded IRQ-NUM= 176
[  993.090567] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]Detect chip WILC1000
[  993.090613] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]loading firmware mchp/wilc1000_wifi_firmware.bin
[  993.090624] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_get_firmware]WLAN firmware: mchp/wilc1000_wifi_firmware.bin
[  993.091009] wilc_handle_isr,>> UNKNOWN_INTERRUPT - 0x00000010
[  993.091701] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Downloading Firmware ...
[  993.091741] wilc_wlan_firmware_downloadDownloading firmware size = 135832
[  993.115154] wilc_wlan_firmware_download Offset = 120380
[  993.121309] wilc_wlan_firmware_download Offset = 135832
[  993.121333] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_firmware_download]Download Succeeded
[  993.121392] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_start_firmware]Starting Firmware ...
[  993.126084] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_start_firmware]Waiting for FW to get ready ...
[  993.629338] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_start_firmware]Firmware start timed out
[  993.629352] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:945] Failed to start firmware
[  993.629358] wilc_sdio mmc1:0001:1: wilc_sdio_disable_interrupt
[  993.629503] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[  993.629508] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinit debug Thread
[  994.109484] wilc_sdio mmc1:0001:1 wlan0: INFO [wlan_deinitialize_threads]Deinitializing Threads
[  994.109595] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:985] WLAN initialization FAILED
[  994.109607] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open:1029] Failed to initialize wilc```

ajaykathat, 2020-07-23:
> ```
> [   10.218375] wilc_sdio mmc1:0001:1 wlan0: INFO [set_power_mgmt]dev [wlan0]
> [   10.218388] wilc_sdio mmc1:0001:1 wlan0: INFO [set_power_mgmt] Power save Enabled= 1 , TimeOut = -1
> [   10.245698] [...]

From the above logs, it appears the mac_close is triggered by HOST during the initialization of the driver. The initialization part looks fine but this mac_close is triggered in between. Please check the causes of mac_close() trigger in the CM3 board.

Please give a try by disabling the PowerSave mode. To disable PS mode, either compile kernel by disabling default PS (CONFIG_CFG80211_DEFAULT_PS) or modify 'wilc_set_power_mgmt' function to always set 'power_mode' variable with WILC_FW_NO_POWERSAVE value.

```
	if (enabled)
		power_mode = WILC_FW_MIN_FAST_PS;
	else
		power_mode = WILC_FW_NO_POWERSAVE;
```
to
```
	power_mode = WILC_FW_NO_POWERSAVE;
```

Andrflor, 2020-07-24:
I tried both options but it didin't end well...
I'm wondering if it could be because of reset and chip_en pin state?

```[  434.691431] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[7a429e69] wlan0
[  434.691437] power up request for already powered up source Wifi
[  434.691441] Device already up. request source is Wifi
[  434.691448] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[7a429e69][0014ce1f]
[  434.691459] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[  434.691465] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[  434.691495] wilc_sdio mmc1:0001:1: SDIO speed: 25000000
[  434.691894] wilc_sdio mmc1:0001:1: wilc_sdio_cmd52..failed, err(-110)
[  434.691904] wilc_sdio mmc1:0001:1: Fail cmd 52, enable csa...
[  434.691912] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:905] Initializing WILC_Wlan FAILED
[  434.691917] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open:1029] Failed to initialize wilc
[  434.692888] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]MAC OPEN[7a429e69] wlan0
[  434.692894] power up request for already powered up source Wifi
[  434.692897] Device already up. request source is Wifi
[  434.692905] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_init_host_int]Host[7a429e69][0014ce1f]
[  434.692913] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mac_open]*** re-init ***
[  434.692919] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_wlan_init]Initializing WILC_Wlan
[  434.692925] wilc_sdio mmc1:0001:1: SDIO speed: 25000000
[  434.693039] wilc_sdio mmc1:0001:1: wilc_sdio_cmd52..failed, err(-110)
[  434.693045] wilc_sdio mmc1:0001:1: Fail cmd 52, enable csa...
[  434.693052] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_wlan_initialize:905] Initializing WILC_Wlan FAILED
[  434.693057] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_mac_open:1029] Failed to initialize wilc
[  434.693082] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693088] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693093] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693099] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693105] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693110] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693116] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693121] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693127] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693132] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693138] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693143] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693148] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693154] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693160] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693165] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693170] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693176] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693182] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693187] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693193] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  434.693198] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_mgmt_frame_register]Return since mac is closed
[  435.594678] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_send_config_pkt:1806] Set Timed out
[  435.594695] wilc_sdio mmc1:0001:1 wlan0: ERR [wilc_set_operation_mode:1858] Failed to set driver handler
[  435.594712] wilc_sdio mmc1:0001:1 wlan0: INFO [wilc_netdev_cleanup]Unregistering netdev 7a429e69
[  435.684908] De-Initializing Locks
[  435.734648] Freeing wiphy
[  435.734669] Module_exit Done.
[  435.734675] at_pwr_dev: deinit
[  435.734920] at_pwr_dev: unregistered
[  435.735050] mmc1: card 0001 removed