linux4wilc/driver #104: Upstream roadmap

jan-kiszka, 2020-09-14:
What's the relationship of this activity and the upstream wilc1000 driver? Are the plans to enhance the upstream version with wilc3000 support? Is there maybe a staging queue for upstream?

ShonkyCH, 2020-09-14:
Probably worth reading the response here: https://github.com/linux4wilc/driver/issues/30

We're told this is the official version, but yes there seems to be semi parallel efforts. So in the two years, no progress seems to have been made wrt to the mainline kernel.

jan-kiszka, 2020-09-14:
wilc1000 come out of staging by now (5.9). Apparently similar people involved for that. So it seems logical to me to ask for the next steps. Would be beneficial for everyone if this driver here can be retired.

tsifb, 2020-09-14:
Interesting. Have you had a chance to diff that version with the the code here? There was quite some divergence a while ago.

jan-kiszka, 2020-09-14:
They are still fairly different, though not completely. It will likely take some effort extract features and fixes from here, that's also why I'm asking for a potentially existing upstream queue with pending changes. The sooner everyone can and does focus on upstream, the better.

tsifb, 2020-09-14:
It would certainly be nice to have 100% of resources going towards fixing issues, providing customer support and improving functionality.

Does upstream use the same firmware as is current in the firmware repo here? This is probably the most critical, being a closed-source binary blob.