linux4wilc/driver #105: iPhone / iOS Authentication Problem

bht-patrick-schneider, 2020-09-19:
Hi everyone,
we are facing a problem with our ATWILC3000 module.
We are running as an AP with hostpad - if an iOS device (iPhone, iPad) tries to connect to this AP with SSID and Password **for the first time** it shows a "password incorrect" error. Re-Entering the password solves the issue.
If the network settings are reset to factory default the issue rises again.
Release Notes for the 15.2 release state: "iPhone intermittently fails to connect to ATWILC SoftAP" but the problem persists with this release and all following.
Is this a known issue, does anybody experience this beside us? Shall we provide more information?
It is a **huge** sales and marketing problem if every new user we show the device to has to reenter the password a second time.
Cheers!

Mateusz-Gwara, 2020-09-25:
This is interesting,

We had a similar problem on Android 8.x and 9.x on Samsung SM-T580 and SM-T510 Tablets.
With the current firmware and drivers we have the case, where tablets which already been connected but lose their connection are unable to reconnect automatically. If you manually tap on the network they connect just fine.
Resetting the WILC3000 chip and reloading the drivers and hostapd solves the issue as well.

Maybe these problems are related?

bht-patrick-schneider, 2020-10-14:
Unfortunately resetting the chip or reloading the driver does not solve the issue on our side.
New information shows the following: if NO OTHER wi-fi network is active the problems get MUCH WORSE. Also Android Phones start to react strange while connecting and the iPhone Problem gets more frequent and sometimes results in complete connection refusal.
Does anyone from the contributors have an idea on this case?
@ajaykathat @AdhamAbozaeid