linux4wilc/driver #114: Impact of changing AP channel in concurrent AP/STA mode

BDP42, 2021-04-27:
We are running the WILC3000 device in concurrent AP/STA mode.

The AP is up and running on a given channel.
The supplicant is running and scanning on a regular basis (once every 2 minutes) for known external APs and automatically connects if one of these is detected.

Due to the fact that the AP and STA have to be on the same channel, we may need to change the channel the AP is using if the supplicant connects to an external AP that is operating on a different channel.

We do this by stopping hostapd, changing the channel in hostapd.conf and restarting hostapd. This is the only way I'm currently aware of changing the channel ( see issue https://github.com/linux4wilc/driver/issues/113 related to this I recently posted).

However stopping hostapd removes wlan0 interface and I need to rmod and then insmod the WILC driver to bring wlan0 back.
This has a consequence on the supplicant since the phy number seems to change every time this happens i.e. phy0 -> phy1 -> phy2....etc. and makes the management of the supplicant connecting to an external AP while keeping the WILC AP up a bit messy.

Has anyone else experienced this kind of issue when using concurrent AP/STA mode?
Is there a better way to do this other than that described here? 

Thanks,
Bruno

ajaykathat, 2021-04-27:
Hi Bruno,

One approach could be to use " iw phy0 interface add wlan0 type __ap" command to add interface after killing hostapd and no need to remove the module(rmod). The sequence should be like stop hostapd, change the channel in hostapd.conf, add interface ("iw phy0 interface add wlan0 type __ap" command) then start hostapd.  To know the current PHY number(wiphy index)  "iw phy" command can be used.

Regards,
Ajay

Regards,
Ajay

BDP42, 2021-04-28:
Thanks @ajaykathat  for your suggestion.
This worked and so no need to rmod/insmod the driver.

closed 2021-04-28 by BDP42