linux4wilc/driver #119: ISSUE: Failed to initialize wilc (linux4sam-2021.04)

davidtekin37, 2021-12-22:
Hello, 
I am facing an issue with the WILC3000 firmware 15.5 directly from the kernel 5.10.50.

I have my SAMA5d27-som1 and my WILC3000 fully working with the version (linux4sam-6.0 WILC 15.2).
But now I want to upgrade to the last version.

I use the WILC3000 in SPI mode, this configuration is defined in my dts file. 
[image]

When I want to run the firmware I am facing an issue, the firmware won't start.
[image]

Anyone does have the same issues? I think there is an error on my device tree file, but I don't know where.

davidtekin37, 2021-12-24:
In order to continue the investigation, I captured the debug of the WILC3000 on the UART serial port. Here is the TX output of the module.
[image]

ajaykathat, 2021-12-30:
Hi @davidtekin37,

From the logs, it seems the FW start is not received by the Host. Could you please probe the IRQ line from the WILC board?  FW sends the interrupt to the host when it starts. 
As it was working earlier please check if there are any changes in the IRQ GPIO configuration.
Also please try with the latest driver from [WILCv15.6](https://github.com/linux4microchip/linux/tree/linux-5.10-mchp/drivers/net/wireless/microchip)
Otherwise, please open the salesforce ticket for this issue and the support team can help.
http://microchipsupport.force.com/