linux4wilc/driver #122: Issues in Wilc3000 integarte with BBB

20023356, 2023-06-18:
I am integrating wilc3000 with BBB using Kernel 5.10 ([Debian: Getting Started with the BeagleBone Black]. This version Pin numbers are different from 4.19 kernel version.
In 5.10 kernel SPI pin numbers are 
gpio-78  (P9_31 [spi1_sclk]   |P9_31               ) out lo 
 gpio-79  (P9_29 [spi1_d0]     |P9_29               ) out lo 
 gpio-80  (P9_30 [spi1_d1]     |P9_30               ) in  lo 
 gpio-81  (P9_28 [spi1_cs0]    |P9_28               ) out lo 
while booting and insmod wilc-spi.ko i got the following errors

ebian@beaglebone:~$ sudo ifconfig wlan0 up
[  996.034137] power up request for already powered up source Wifi
[  996.044941] WILC_SPI spi1.0: Failed cmd response, cmd (c4), resp (00)
[  996.059352] WILC_SPI spi1.0: Failed internal read cmd...
[  996.068919] WILC_SPI spi1.0: Failed cmd response, cmd (cf), resp (00)
[  996.083265] WILC_SPI spi1.0: Failed cmd reset
[  996.091484] WILC_SPI spi1.0: Reset and retry 10 24
-
-
[  997.003210] WILC_SPI spi1.0: Reset and retry 1 24
[  997.014504] WILC_SPI spi1.0: Failed internal read protocol...
[  997.023888] WILC_SPI spi1.0 wlan0: ERR [wilc_wlan_initialize:905] Initializing WILC_Wlan FAILED
[  997.036400] WILC_SPI spi1.0 wlan0: ERR [wilc_mac_open:1029] Failed to initialize wilc
SIOCSIFFLAGS: Input/output error

Please help me to resolve this issue and what could be the issue.
Regards
Ravi V

ajaykathat, 2023-06-19:
Apart from SPI pins connection, the WILC's RESET_N and CHIP_EN pins should be connected with host GPIO's for the power-up sequence. Assuming the SPI is correctly configured, this failure could be because the WILC power-up sequence is not triggered by the host. The details about the power sequence are captured in "4.1 ATWILC Power Control" section of wilc1000/3000 porting guide which can be referred from the [link](http://ww1.microchip.com/downloads/en/DeviceDoc/ATWILC1000-ATWILC3000-ATWILC-Devices-Linux-Porting-Guide-User-Guide-DS70005329C.pdf). Host, BBB in this case, GPIO's should be connect to WILC's RESET_N and CHIP_EN pin for chip power-up sequence. Please try connecting the host GPIO's to RESET_N and CHIP_EN if it helps to overcome this issue.

20023356, 2023-06-21:
Thanks Ajay, 
still am facing the same issue. could you please help me to resolve this issue
Also pls find the conf file i have addded

dhcpd.conf:

option domain-name "domain.name";
option domain-name-servers 192.168.0.1;
default-lease-time 1440;
max-lease-time 1440;
authoritative;

subnet 192.168.0.1 netmask 255.255.255.0 {
        option routers                  192.168.0.1;
        option subnet-mask              255.255.255.0;
        option domain-search            "domain.name";
        option domain-name-servers      192.168.0.1;
        range   192.168.0.10   192.168.0.100;
        range   192.168.0.110   192.168.0.200;
}

wilc_wpa_supplicant.conf

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1

network={
ssid="D-Link_HBIO"
psk="hbio2ltts"
proto=RSN
key_mgmt=WPA-PSK
pairwise=CCMP
auth_alg=OPEN
}

wilc_hostapd_wpa.conf:

interface=wlan0
driver=nl80211
ctrl_interface=/var/run/hostapd
ssid=wilc_OpenAP
dtim_period=2
beacon_int=100
channel=7
hw_mode=g
max_num_sta=8
ap_max_inactivity=300 

wilc_hostapd_wep.conf:

interface=wlan0
driver=nl80211
ctrl_interface=/var/run/hostapd
ssid=wilc_WepAP
dtim_period=2
beacon_int=100
channel=7
hw_mode=g
max_num_sta=8
ap_max_inactivity=300
ieee80211n=1
auth_algs=1

######### WEP ###########
wep_default_key=0
wep_key0=1234567890
wep_key1="vwxyz"
wep_key2=0102030405060708090a0b0c0d
wep_key3=".2.4.6.8.0.23"
wep_key_len_broadcast=5
wep_key_len_unicast=5
wep_rekey_period=300

/etc/network/interfaces:

allow-hotplug wlan0
auto wlan0
iface wlan0 inet manual
wpa-roam /etc/wpa_supplicant/wpa_supplicant.conf
iface default inet dhcp

20023356, 2023-06-21:
> CAUTION: This email is from an external source. Please do not open attachments or click links from an unknown or suspicious origin. Phishing attempts can be reported by using the report message button in Outlook or sending them as an attachment to ***@***.******@***.***>. Thank you

Thanks Ajay

Still the problem is not resolved. I have shared the details in git hub.

So, could you please give me suggestion to resolve this issue. it would you be very much appreciate.

Regards
Ravi V

L&T Technology Services Ltd

www.LTTS.com

L&T Technology Services Limited (LTTS) is committed to safeguard your privacy. Read the appropriate privacy notice<https://www.ltts.com/privacy-policies-notices> applicable to you to know more about how LTTS collects, uses and stores your personal data. Please visit our privacy policy<https://www.ltts.com/data-privacy-policy> available on our website www.Ltts.com for more information about our commitment towards data privacy under applicable data protection laws. This email may contain confidential or privileged information for the intended recipient(s). If you are not the intended recipient, please do not use or disseminate the information, notify the sender and delete it from your system.

ajaykathat, 2023-06-21:
It looks like the first SPI command from the host to the WILC has failed. First, check if the SPI bus communication is working fine. Either validate the Device Tree node entry for BBB SPI device or, If possible, use the SPI bus analyzer to probe the Host(BBB) SPI pins to check whether the command is going out correctly. 
After the chip ID read is successful, the firmware gets downloaded to the WILC and, then the actual WLAN functionality starts.  

Since the first command to read the chip ID failed, just using "config wlan0  up" should be enough to test this.  
Short procedure to test:
$load driver if its loadable module
$ifconfig wlan0 up

If you continue to face issue intergrating WILC with BBB, please open a ticket on Microchip's salesforce portal http://microchipsupport.force.com/ so that respective team can help further.